prusa3d/PrusaSlicer #609: Suggestion for Fix STL process

BirkBinnard, 2017-12-05:
### Version
1.38.3

### Behavior
* _Describe the problem_
Too many steps required to fix an STL file and have it sliced and saved as a repaired STL.

* _Steps needed to reproduce the problem_
1. Click Repaire STL File and save it as an OBJ file
2.  Load OBJ file
3. Save as STL file (I always write over the original one to maintain the same name)
4. Delete OBJ file
5. Load STL file

* _Desired Results_
1. Click Repair STL file and have it saved as either/both STL and/or OBJ file
(In other words, after repairing, automatically engage the STL conversion routine and give option to save one or both versions.)

_Is this a new feature request?_
Yes

The Repair function is one of the most useful features; I use it all the time.  I'd be happy to provide the Save OBJ/STL dialog box, but I only code in Visual Basic.

bubnikv, 2018-08-02:
I wonder what do you use the repair function for?

Actually I am considering the repair function in Slic3r pretty poor, so I am inclined to refuse your request. 

We have integrated the fabulous Windows 10 Netfabb API recently though.

rtyr, 2020-07-29:
No response. Closing.

closed 2020-07-29 by rtyr

BirkBinnard, 2020-07-29:
Sorry I missed your response. Thanks. 

⁣-----
Birk Binnard ​