prusa3d/PrusaSlicer #610: wait for user...

totalitarian, 2017-12-06:
Why do multi colour prints pause occasionally with the prompt on the lcd that says "wait for user...."

Tried to do a multi color print last night and the print head stayed parked over the priming tower for hours on end. I tried to resume the print in the morning but the nozzle had melted the prime tower after being parked.

bubnikv, 2017-12-06:
I think we can close the Slic3r issue, right? It seems you either have a firmware or a hardware issue.

totalitarian, 2017-12-06:
Whatcha the gcode that produces  the wait command. Let me recheck my sliced file before we close the slic3r issue please.

totalitarian, 2017-12-06:
Just for reference here is my gcode. If you can't spot anything causing the pause in there, feel free to close this issue.
[Buldozer.zip](https://github.com/prusa3d/Slic3r/files/1536596/Buldozer.zip)

bubnikv, 2018-08-02:
We hope this was a firmware issue and it has been fixed. Closing.

closed 2018-08-02 by bubnikv