prusa3d/PrusaSlicer #611: The problem is the poor quality of the first layer from the bottom up

TomRookie, 2017-12-06:
### Version
1.37.2.prusa3d.win64

### Operating system type + version
Windows 8.1

### Behavior
The problem is the poor quality of the first layer from the bottom up.
When I use support, the first layer is scattered. 
Is it possible to change the density of lines?
Or how can I make it better?

[image]
[test.zip](https://github.com/prusa3d/Slic3r/files/1535735/test.zip)

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

Sebastianv650, 2017-12-06:
You might have a look into #102 . It's not straight forward, but there is a way to fool Slic3rs way of support generation, resulting in better quality. See last post for pictures.
Your result may vary and I think you would have to play with some settings for best results.

But there are more things that need to be changed to get the desired 3D geometry on the print when using supports. It's on my list to open issues for that when I get some time, up to now my problem is how to describe the changes I would suggest in a way that is understandable for 3rd people.. But I'm not in a hurry because I don't think the support logic will be changed soon.

sepotvin, 2017-12-13:
I found that if I set the z contact distance to 0.05 mm and the min layer height to 0.05 (from 0.07 mm default) I get much better result than with the default settings although it's still not perfect. Take note that it might not work for your filament as bubnikv indicated that Prusa had issues with 0.05mm min layer height.

bubnikv, 2021-03-18:
To improve quality of the object over support, we have added a new settings: "thick bridges". If enabled (that is by default), PrusaSlicer behaves as before. If disabled, bridges are printed as in any other slicer: The extrusion rate is given by the normal layer height x extrusion width * bridge flow ratio. 

[image]

Implemented with ceea9de8b8b1d4b35bdd1180051033680c405ba1 f01f02154c617cc24835475da419ed6322f2ceec

closed 2021-03-18 by bubnikv