prusa3d/PrusaSlicer #612: Suggestions for support logic changes

Sebastianv650, 2017-12-06:
This is meant to be a follow up and extension of #102, triggered by #611. Only suggenstions, but I think they would work great.

1. At the moment, each support will lower the 3D print result geometry compared to the 3D imput file. To get a 3D print that has as-designed dimensions, the top support surface layer has to be at input-file height level and the next printed layer on top of support has to be shifted upwards depending on Contact z distance:
[image]

2. (Covered by #102) Contact z-distance is defined as a gap additional to zero squish layer height (=nozzle diameter). This way, the first printed layer over support will be wobbly on arcs as it is not supported until the line sags ontu the support layer. Additional, the squish of the second printed layer of support is also wrong, as it's layer height will always be layer height + Contact z-distance. That's why it's often possible to pull of parts from the first printed layer from the finished print.
Proposal: Define a Contact z-squish. For very sticky materials where a real gap as defined today would be necessary (I know none?), this wouldn't be sufficient as sqush can't be >1 (=extrusion in free air).
To keep a possibility for this and to avoid confusion from users used to have a setting in Unit [mm], it would be possible to define a z-distance where 0 would mean a squish as on every layer height and a logic compareable to current one when a distance > (nozzle diameter - layer height) is entered. Big drawback: A propper z-distance to remove supports at 0.2mm layer height may lead to unremovable supports when layer height is set to 0.1 and vice versa as the resulting squish is variable.
[image]

3. In every case, the extrusion width and line distance should result in a dense first layer also over support. This might be wrong when using a squish > 1, meaning a real gap as described above.

Feel free to comment and ask if something is not clear :)

bubnikv, 2017-12-07:
1) I agree. This issue applies to the bridges as well.
2) I am not sure I understand everything you wrote.
3) Everybody wants that and you can achieve that over soluble supports. There is no way you can achieve that over the same-material support.

You made some good points. For example, the 2nd object layer over supports should likely be squished hard to the 1st object layer printed over the support interface to be geometrically correct. It has to be tried, though it is non-trivial to implement as it does not fit the Slic3r architecture easily. One may extrude the 2nd object layer over supports thicker though. I bet no slicer does this.

I don't quite agree with your ideas of squishing the 1st object layer to the support interface though. The squishing is really a trade-off between the bottom surface quality and removability. You seem to only consider flat bottom, easily accessible object surfaces. But if you applied your squishing to some less easily accessible spots, you would never be able to remove those supports. Then somebody would certainly come with the idea of a manual editing tool or an automatic deduction of accessibility algorithm. That is certainly all doable, but we need to start with something manageable in reasonable time.

It is certain, that the support has a lot to be desired, and that a lot of experimenting is needed to improve it greatly.

One idea I had is to squish the innermost perimeter more against the support interface as an exception, so the shape of the perimeter would be well defined, and the infill would have something to stick to. As only one perimeter would be fused to the support interface, it would still be possible to remove it from the supports quite easily, contrary to your proposals to squish every line.

Other idea applicable to the bridges as well, is to lay the infill in an even-odd order: To lay the even lines first into the air, and then to squish the odd lines into the gaps. This would lead to a slightly wobbly bottom surface, but this wobbly bottom surface would be filled without gaps. Whether it would work is hard to say without somebody implementing it and toying with it.

I think the single biggest request on Slic3r from most of our users are the user editable supports, as most of our customers are printing complex objects like statues, game character models and similar, where the automatic support generator does not place the supports where it should in many cases.

bubnikv, 2021-03-18:
To improve quality of the object over support, we have added a new settings: "thick bridges". If enabled (that is by default), PrusaSlicer behaves as before. If disabled, bridges are printed as in any other slicer: The extrusion rate is given by the normal layer height x extrusion width * bridge flow ratio. 

[image]

Implemented with ceea9de8b8b1d4b35bdd1180051033680c405ba1 f01f02154c617cc24835475da419ed6322f2ceec

bubnikv, 2021-09-23:
Implemented by the "thick bridges" configuration value.

closed 2021-09-23 by bubnikv