prusa3d/PrusaSlicer #613: Printer settings  >  Start G-code and End G-code - Not working properly

fractalbrain, 2017-12-07:
Hiya!!  Thanks for all the hard work!! -ej

### Version
1.38.3-prusa3d

### Operating system type + version
MacOS 10.13.1

### Behavior 1
When "Printer settings">"Custom G-code">"Start G-code" had comments at the beginning, the outputted gcode had a combination of my g-code and other g-code.  That resulted in the bed homing, and then continually trying to travel in the opposite direction past the max, resulting in the belt skipping.

Removing the comments caused everything to work properly.

 ### Behavior 2
The "Printer settings">"Custom G-code">"End G-code" is not working properly.  The first several lines get incorporated into the g-code, but then the rest is replaced by what looks like a default.

Here is what it should look like:
`G1 E-0.5 F200		; retract the filament a bit before lifting the nozzle, to release some of the pressure.
G1 Z+40 E-5 F3000	; move Z up a bit (if possible) and retract filament even more.  That’s 5.5mm to offset later.
G90                          	  ; absolute positioning
G1 X145 Y250               ; move to cooling and cleaning position
M84                            ; steppers off
G90                            ; absolute positioning
G4 S30					; Wait while hot end cools
M107						; Turn off fan
M117 TAZ Ready.
`

...but here is what is actually exported:
`
M400 				; Wait for buffer to clear - ej
M106 S255			; Turn on fan full
M104 S0                        ; hotend off
M140 S0                        ; heated bed heater off (if you have it)
M106 P0 S255 		; hot end cooling fan max
G91                           	 ; relative positioning
G1 E-0.5 F200		; retract the filament a bit before lifting the nozzle, to release some of the pressure.
G1 Z+40 E-5 F3000	; move Z up a bit (if possible) and retract filament even more.  That
; filament used = 6485.5mm (42.2cm3)
; filament used = 52.8
; filament cost = 2.3
...
`

#### STL/Config (.ZIP) where problem occurs
[config.ini.zip](https://github.com/prusa3d/Slic3r/files/1540213/config.ini.zip)

bubnikv, 2017-12-09:
I believe this is a duplicate of https://github.com/prusa3d/Slic3r/issues/600
If you download the current 1.38.3 dmg and write the G-code comments from scratch, the apostrophe character will not be "beautified" by OSX by translating it to an UTF-8 sequence, and the G-code will work.

The 1.38.4 will process the UTF-8 sequences in the comments as well.

bubnikv, 2017-12-14:
Closing, fixed with #600.

closed 2017-12-14 by bubnikv