prusa3d/PrusaSlicer #614: Slic3r 1.38.3 gcodes not building up bed temperature

jok250, 2017-12-08:
### Version
Slic3r Prusa Edition 1.38.3-prusa3d-win64

### Operating system type + version
Win 8.1 x64

### Behavior
Just updated this afternoon to the new 1.38.3 and noted that the bed pre-heating didn't worked for me any more... looking into the gcode, apparently there's one line missing:

[image]

#### STL/Config (.ZIP) where problem occurs
[Slic3r_config_bundle.zip](https://github.com/prusa3d/Slic3r/files/1544070/Slic3r_config_bundle.zip)

bubnikv, 2017-12-09:
Thanks for your findings. This is a regression error introduced in Slic3r 1.38.2, it has been fixed with https://github.com/prusa3d/Slic3r/commit/8509e4b5f5ca87b5408b878d524051f7324c6908.
The fix will appear in Slic3r 1.38.4

bubnikv, 2017-12-14:
Implemented in 1.38.4. Closing.

closed 2017-12-14 by bubnikv