prusa3d/PrusaSlicer #615: Output filename generation based on variables broken

rongith, 2017-12-09:
### Version
1.38.3

### Operating system type + version
Windows 10 64bits

### Behavior
Output filename format, in my case: [input_filename_base]-[P][print_preset]-[F][filament_preset]-[P][printer_preset]-[version]-[timestamp].gcode
Don't give a proper filename anymore.

bubnikv, 2017-12-09:
Slic3r 1.38.4 will emit an error report, that a variable named P does not exist. (that is for the [P] sequence).
I know the old Slic3r just ignored the non-existent variables, and you used this behavior. The new Slic3r will report it as an error.

rongith, 2017-12-11:
The following output format filename doen't work too:
[input_filename_base]-(P)[print_preset]-(F)[filament_preset]-(P)[printer_preset]-[version]-[timestamp].gcode
Per exemple 1.37.2 gives:
Rectangle-250x210 - Defaut-(P)0.20-020pc-vmax-(F)Prusa PET-(P)PMK2 - Echelle-R0.8-1.37.2-prusa3d-win64-20171211-014804

bubnikv, 2017-12-11:
I have uploaded a continuous build, which has the placeholder parser error reporting significantly improved. Would you please test with that? I just tested with your new template 
[input_filename_base]-(P)[print_preset]-(F)[filament_preset]-(P)[printer_preset]-[version]-[timestamp].gcode
and it seem to work, it produces
20mmbox-(P)0.15mm OPTIMAL-(F)ColorFabb HT 1.75mm-(P)Original Prusa i3 MK2 MultiMaterial-1-20171211-092405.gcode

[Slic3r-1.38.3.21-prusa3d-win64-full-gae5863f-201712102228.zip](https://github.com/prusa3d/Slic3r/files/1547007/Slic3r-1.38.3.21-prusa3d-win64-full-gae5863f-201712102228.zip)

bubnikv, 2017-12-11:
I found a bug in the "version" variable. I will post another continuous build soon.

bubnikv, 2017-12-11:
Here is an updated build.

[Slic3r-1.38.3.22-prusa3d-win64-full-g1938828-201712110832.zip](https://github.com/prusa3d/Slic3r/files/1547090/Slic3r-1.38.3.22-prusa3d-win64-full-g1938828-201712110832.zip)

This is the error message you receive from this build on your original template:
[image]

It is not ideal as the ^ arrow should point to the variable, but due to variable width font it does not. I shall fix that as well.

rongith, 2017-12-11:
Nice!
I got now Rectangle-250x210 - Defaut-(P)0.20-020pc-vmax-(F)Prusa PET-(P)PMK2 - Echelle-R0.8-1.38.3.22-prusa3d-win64-20171211-141837.gcode

closed 2017-12-11 by rongith