prusa3d/PrusaSlicer #616: 1.38 won't start on Win10

RudolphRiedel, 2017-12-09:
### Version
1.38.2
1.38.3
### Operating system type + version
Windows 10 Home 64-Bit
### Behavior
Right now while I am testing it starting slic3r.exe from
Slic3r-1.38.2-prusa3d-win64-full-201711292009.zip
Slic3r-1.38.3-prusa3d-win64-full-201712012144.zip
does nothing, it just silently fails.
Even when trying to start it as admin.

When I copy the files from a 1.38.3 archive to a directory from which I started 1.37,
a blue Windows Defender screen is popping up telling me that the launch of an unknown app has been prevented.
And when I agree to start it regardless it again just silently fails to start.

ir-fuel, 2017-12-09:
Did you try starting it on the command line? It's maybe the same error as on other platforms

RudolphRiedel, 2017-12-09:
Well no, I did not try that before.
Now I did and while "slic3r.exe" immediately returns to the command-prompt with no output whatsoever. "slic3r-console.exe" hangs for a while then windows pops up a window telling me that it had to abort it's execution since it was not responding.
And "slic3r-nocosole.exe" also just returns to the command prompt with no output.

bubnikv, 2017-12-09:
This is weird. May it be blocked by some antivirus or a Windows Defender?

Did you install the previous Slic3r with the Prusa3D windows drivers installer, or did you just unpack the zip file?

RudolphRiedel, 2017-12-09:
There is an installer?
I just unpacked the .zip files from here: https://github.com/prusa3d/Slic3r/releases
1.37.2 works that way and I just unpacked it again to a different directory.
1.38.2 and 1.38.3 just trigger a mouse-pointer animation for a few seconds and silently fail to start.

bubnikv, 2017-12-09:
There is a big driver package on prusa3d.com, which contains printer drivers, example objects, pronterface and prusa control all bundled into a single install. You don't need that, I just needed to know.

Would you please zip and attach the content of your slic3r configuration directory? I reworked the configuration layer into C++ and refactored it, so I may have introduced some bugs.

You will find your slic3r profiles here:
"C:\Users\your_user_name\AppData\Roaming\Slic3r" or "C:\Documents and Settings\your_user_name\Application Data\Slic3r"

RudolphRiedel, 2017-12-09:
Attached is that whole directory, I just closed 1.37.2 to zip that.
I admit that it is messy, I merely started usind slic3r.
[Slic3r.zip](https://github.com/prusa3d/Slic3r/files/1544860/Slic3r.zip)

What I noticed is that while 1.38.3 does not seem to do anything, it modifies the slic3r.ini when I start it.

bubnikv, 2017-12-09:
Your print\My Settings.ini is a printer profile, it should not be inside the print directory. I bet Slic3r will start, if you remove that particular file.

I should improve Slic3r though, so it will not crash.

bubnikv, 2017-12-09:
Fixed with https://github.com/prusa3d/Slic3r/commit/9a80ff57b26b41bf1e4ad275ccd8416cd0d567b0
It will be part of the next build.

closed 2017-12-09 by bubnikv

RudolphRiedel, 2017-12-09:
I removed that file and neither 1.38.2 or 1.38.4 do start.
Okay, I went one step further and renamed that directory and 1.38.3 does start.

After going thru the basic configuration, there is a new "My Settings.ini" in "print". And there is one in "filament" and in "printer" - so these are auto-generated by Slic3r itself.

Okay, I re-activated that first directory and removed the other "My Settings.ini" files and now 1.38.3 as well as 1.38.2 start.

My guess is, something got broken in that files between 1.37 and 1.38.

bubnikv, 2017-12-10:
> My guess is, something got broken in that files between 1.37 and 1.38.

It may have got broken much earlier, but up to 1.38 the Slic3r did not mind.