prusa3d/PrusaSlicer #618: Unable to re-slice and save after removing model copies

Popeguy, 2017-12-09:
### Version
1.37.2

### Operating system type + version
Windows 10 Pro x64
Version 10.0.16299 compiled 16299

### Behavior
After slicing, saving gcode and removing model copies I am unable to slice/save gcode: https://imgur.com/8Jc0yzg
Program restart is required.

1) Add the model (.stl file)
2) Righ click, and rescale to size (uniformly)
3) Click the plus sign ("copy model" from the top menu) 5 times
4) Click "Arrange" from the top menu
5) Click Slice, then click Export g-code and save it somewhere
6) Before that G-code is saved (green bar still visible), click minus sign on top menu (REMOVE COPY)

Confirmed above works with my friend's slicer (same version, non-mk3 profile).

#### STL/Config (.ZIP) where problem occurs
Works with every .stl file.

bubnikv, 2018-03-15:
@FidelCapo confirmed this bug on Windows with 1.37.2 and with the latest integration build 1.40.0-alpha.249

There is an issue with the cancellation of the slicing core, which is still partially written in Perl. I suppose this issue will go away in 1.41, when the complete slicing backend will be rewritten into C++.

bubnikv, 2018-12-11:
The code responsible for your issue was rewritten from Perl to C++, also the background processing should now work reliably. Please retest with 1.42 alpha once it is released this or the next week. Closing.

closed 2018-12-11 by bubnikv