prusa3d/PrusaSlicer #619: Slic3r pe constantly freezing on Intel HD graphics

Patrik-svobodik, 2017-12-10:
##Version
1.37.2-prusa3d-win64_

Operating system type + version
w10 x64

Behavior
*Constantly freezing when trying to view an stl (sometimes even without stl)
 occurs after start

bubnikv, 2017-12-10:
This is an evergreen, caused by wrong OpenGL driver installed by the Windows updater. The Redmond guys should talk to the guys at Santa Clara more often.

#268
#233

peterjhl, 2017-12-10:
Perhaps it's an evregreen.
Therefore, there is still one thing that should be corrected.
For me, the program is completely useless.
Even though I install the latest Intel update, right from the source.

Win 10 pro - Intel HD 4000 - 10.18.10.4653

bubnikv, 2017-12-10:
> Perhaps it's an evregreen.
> Therefore, there is still one thing that should be corrected.

Please, Microsoft and Intel, please, fix it.

I don't know any other workaround, sorry.

> For me, the program is completely useless.
> Even though I install the latest Intel update, right from the source.

I am sorry to hear that, but I can assure you, that every other case I
heard of was solved by the driver installation. And there were many since
the beginning of this year.

Please read the two issues thoroughly. If you run the Intel auto install
tool, you really need to restart the machine when it asks you to restart
it, otherwise the tool will install the incorrect driver again.

peterjhl, 2017-12-10:
> Please read the two issues thoroughly. If you run the Intel auto install
tool, you really need to restart the machine when it asks you to restart
it, otherwise the tool will install the incorrect driver again.

I know, I'm not completely green behind a PC. ;-)
I just need to install and reboot several times, and bingo then it works.
Just until I start PC up again the next day, then we'll be back again.

Currently, I either use Linux version or sli3er original. Have really given up on the Prusa Sli3er PE win edition for now.

> By the way.
slic3r-mswin-x64-1-2-9a-stable
works Execellent

bubnikv, 2017-12-10:
> Just until I start PC up again the next day, then we'll be back again.

That may happen, but not so often as it seems to.

> slic3r-mswin-x64-1-2-9a-stable
> works Execellent

This is a 2 years old install, which contains critical bugs. Good luck with that.
As you are satisfied, we may close this issue, right?

peterjhl, 2017-12-10:
> This is a 2 years old install, which contains critical bugs. Good luck with that.

What a great support !!!
That is why I respond to this issue.

> As you are satisfied, we may close this issue, right?

I have not really indicated that I am satisfied.
It's not up to me to decide to close this issue. Since it is not me who has reported issue.

bubnikv, 2017-12-11:
@Patrik-svobodik were you able to solve your issue by upgrading the graphics card driver?

Patrik-svobodik, 2017-12-11:
No, I was not able to fix the issue with drivers update/reinstall, my shitty pc does not like the slic3r, prusacontrol is just fine, but limitting for me

bubnikv, 2017-12-11:
Are you able to open the preferences dialog, or is it impossibly slow?

I am thinking of adding a "compatible" mode using OpenGL 1.2 only.

Patrik-svobodik, 2017-12-11:
Maybe, how should I do it? How should I enable the OpenGL 1.2?

Patrik-svobodik, 2017-12-11:
[image]

Patrik-svobodik, 2017-12-11:
I can get into it but it is REALLLLLLLLLLLLLLLY fucking slow!!!!! I dont wanna investigate to this machine because it is for 3d printing only but the next machine for 3d printer would be with amd graphics, this bitch is unbelievable slow and buggy with intel. I know it does not happen with amd graphics, I tried it

Patrik-svobodik, 2017-12-11:
[image]
the preview looks like this, and if it loads it does not respond

bubnikv, 2017-12-11:
Please try this build, check the new "use legacy OpenGL" checkbox on the preferences page, close the application and restart. Then it should work much better, though you should get a better performance with the right OpenGL driver. BTW, I am developing on an intel HD Graphics.

[Slic3r-1.38.3.24-prusa3d-win64-full-g743fc9d-201712111702.zip](https://github.com/prusa3d/Slic3r/files/1548785/Slic3r-1.38.3.24-prusa3d-win64-full-g743fc9d-201712111702.zip)

Patrik-svobodik, 2017-12-11:
Nope, not working

Patrik-svobodik, 2017-12-11:
It kinda works when I am in the preview tab but the 3D tab just does not respond

bubnikv, 2017-12-12:
You may try the following, which worked for one of my colleges:

The intel updater tool
http://www.intel.com/content/www/us/en/ ... etect.html

scans your computer and detects all Intel hardware. It then installs the INF files for each of the Intel hardware, which informs the operating system of the particular hardware devices installed. Before the automatic Intel updater tool there was another tool, which did exactly that:

http://www.intel.com/content/www/us/en/ ... 05835.html

After the INF files are updated, the graphics drivers should be updated. If the 1st step is skipped, a wrong driver may be installed for your computer.

Frankly, I cannot say I understand it throughly, it seems like the Intel updater tool is in a need of improvement as it allows the user to skip and shuffle the installation steps, which must not be skipped or shuffled.

bubnikv, 2017-12-12:
Sorry, the links above are broken. You may find more here:
https://shop.prusa3d.com/forum/software-f13/slic3r-mk2-broken-after-plater-gt-gt-dlp-projector-t3997.html#p29858

bubnikv, 2017-12-12:
Would you please try this continuous build with the new "use legacy OpenGL" enabled? The attached build disables multi sample anti aliasing in addition, which may be the culprit of your issues.

[Slic3r-1.38.3.26-prusa3d-win64-full-g673e98b-201712121949.zip](https://github.com/prusa3d/Slic3r/files/1553147/Slic3r-1.38.3.26-prusa3d-win64-full-g673e98b-201712121949.zip)

Patrik-svobodik, 2017-12-17:
slightly better

Patrik-svobodik, 2017-12-19:
I mean it sometimes works, but most times it doesnt

Patrik-svobodik, 2017-12-25:
Now I don't have to try to fix it because on my new Mac it does not happen. Thanks for trying to fix it. And happy new year.

closed 2017-12-25 by Patrik-svobodik

bubnikv, 2018-02-22:
If you have this issue, would you please re-test with the latest beta build and report the results?  Thank you.

https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1-beta1

The fix consists of removing the OpenGL glFlush() calls. The glFlush() call was superfluous, but it certainly should not lead to such adverse effects. Removal of glFlush() calls certainly solved the issue on one laptop, so let's hope it will fix all the other issues or at least most of them.

Thanks @roesel for providing a laptop for testing, it was a blessing to us.