prusa3d/PrusaSlicer #621: Fails to start on Mac

rambopierce, 2017-12-10:
### Version
Slic3r 1.38.3 - Prusa Edition

OS X 10.11.6

### Behavior  (Previous versions of SLIC3r do start on this Mac system)
Aborts during startup (something to do with Perl) with the following report:

Process:               perl5.22.0 [755]
Path:                  /Volumes/VOLUME/Slic3r.app/Contents/MacOS/perl5.22.0
Identifier:            com.prusa3d.slic3r/
Version:               Slic3r 1.38.3-prusa3d (1.38.3-prusa3d)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           perl5.22.0 [755]
User ID:               501

Date/Time:             2017-12-10 13:11:33.297 -0600
OS Version:            Mac OS X 10.11.6 (15G17023)
Report Version:        11
Anonymous UUID:        69219C37-6A35-0A40-4B0F-C66F6AF00C38

Sleep/Wake UUID:       20625C18-32D2-4391-A4CE-5CDC020FC0F5

Time Awake Since Boot: 1600 seconds
Time Since Wake:       500 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libpng16.16.dylib             	0x0000000101ffeb90 png_create_png_struct + 315
1   libpng16.16.dylib             	0x00000001020064a4 png_create_read_struct_2 + 19
2   libpng16.16.dylib             	0x000000010200648b png_create_read_struct + 27
3   libwx_osx_cocoau_core-3.0.dylib	0x0000000101a5b25a wxPNGHandler::LoadFile(wxImage*, wxInputStream&, bool, int) + 106
4   libwx_osx_cocoau_core-3.0.dylib	0x0000000101a47bbb wxImage::DoLoad(wxImageHandler&, wxInputStream&, int) + 251
5   libwx_osx_cocoau_core-3.0.dylib	0x0000000101a48026 wxImage::LoadFile(wxInputStream&, wxBitmapType, int) + 182
6   libwx_osx_cocoau_core-3.0.dylib	0x0000000101a4553a wxImage::LoadFile(wxString const&, wxBitmapType, int) + 154
7   libwx_osx_cocoau_core-3.0.dylib	0x0000000101904e15 wxBitmap::LoadFile(wxString const&, wxBitmapType) + 661
8   XS.bundle                     	0x000000010125c20a Slic3r::PresetCollection::load_bitmap_default(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 122
9   XS.bundle                     	0x000000010126d89d Slic3r::PresetBundle::PresetBundle() + 1197
10  XS.bundle                     	0x0000000100fe6318 XS_Slic3r__GUI__PresetBundle_new(interpreter*, cv*) + 200
11  perl5.22.0                    	0x0000000100b94d53 Perl_pp_entersub + 2659
12  perl5.22.0                    	0x0000000100b8c786 Perl_runops_standard + 22
13  perl5.22.0                    	0x0000000100b0307e Perl_call_sv + 1054
14  Wx.bundle                     	0x00000001021643e0 XS_Wx___App_Start(interpreter*, cv*) + 320
15  perl5.22.0                    	0x0000000100b94d53 Perl_pp_entersub + 2659
16  perl5.22.0                    	0x0000000100b8c786 Perl_runops_standard + 22
17  perl5.22.0                    	0x0000000100b028ce perl_run + 942
18  perl5.22.0                    	0x0000000100add21b main + 155
19  libdyld.dylib                 	0x00007fff936505ad start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff8cb7d5e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff94f73578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff94f71341 start_wqthread + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8cb7defa kevent_qos + 10
1   libdispatch.dylib             	0x00007fff9906b165 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff9906adcd _dispatch_mgr_thread + 52

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8cb7d5e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff94f73578 _pthread_wqthread + 1283
[... 334 lines cut ...]
TOTAL                            489.5M      632 

Model: MacBookPro5,3, BootROM MBP53.00AC.B03, 2 processors, Intel Core 2 Duo, 2.8 GHz, 4 GB, SMC 1.48f2
Graphics: NVIDIA GeForce 9400M, NVIDIA GeForce 9400M, PCI, 256 MB
Graphics: NVIDIA GeForce 9600M GT, NVIDIA GeForce 9600M GT, PCIe, 512 MB
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1067 MHz, 0x80CE, 0x4D34373142353637334548312D4346382020
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1067 MHz, 0x80CE, 0x4D34373142353637334548312D4346382020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x8D), Broadcom BCM43xx 1.0 (5.106.98.101.28)
Bluetooth: Version 4.4.6f2, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Samsung SSD 850 EVO 500GB, 500.11 GB
Serial ATA Device: MATSHITADVD-R   UJ-868
USB Device: USB 2.0 Bus
USB Device: Card Reader
USB Device: USB 2.0 Bus
USB Device: Built-in iSight
USB Device: USB Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: IR Receiver
USB Device: USB Receiver
USB Device: USB Bus
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus:

bubnikv, 2017-12-10:
Isn't it similar to https://github.com/prusa3d/Slic3r/issues/603 or https://github.com/prusa3d/Slic3r/issues/616 ?

Will Slic3r start, if you rename the directory ~/Library/Application\ Support/Slic3r/ to some backup or delete it and start from scratch?

Would you please first zip the directory  ~/Library/Application\ Support/Slic3r/ and attach it to this issue? I suppose there is some issue with the presets. Thank you.

rambopierce, 2017-12-10:
Thanks for your fast support.  I don't think it is the same as either of the issues above.  Moving the settings directory out of the way did not make any difference.  I am on an older Macbook Pro and the latest version of OS X supported on this hardware.  Wonder could it be an update to Perl is needed?  Note, Version 1.37.2 starts just fine.

fiveangle, 2017-12-11:
Works fine here:

> System Software Overview:
> 
>   System Version:	macOS 10.13.1 (17B1003)
>   Kernel Version:	Darwin 17.2.0
>   Time since boot:	6 days 18:55

rambopierce, 2017-12-11:
Tried the provided continuous build and results in similar failure.  Here is the update in case there is any difference:

Process:               perl5.22.0 [1149]
Path:                  /Applications/Slic3r.app/Contents/MacOS/perl5.22.0
Identifier:            com.prusa3d.slic3r/
Version:               Slic3r 1.38.3.21-prusa3d (1.38.3.21-prusa3d)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           perl5.22.0 [1149]
User ID:               501

Date/Time:             2017-12-11 08:02:42.743 -0600
OS Version:            Mac OS X 10.11.6 (15G17023)
Report Version:        11
Anonymous UUID:        69219C37-6A35-0A40-4B0F-C66F6AF00C38

Sleep/Wake UUID:       89C74879-B66F-4D4F-8BB1-CE753A5D9034

Time Awake Since Boot: 11000 seconds
Time Since Wake:       320 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libpng16.16.dylib             	0x0000000108e5cb90 png_create_png_struct + 315
1   libpng16.16.dylib             	0x0000000108e644a4 png_create_read_struct_2 + 19
2   libpng16.16.dylib             	0x0000000108e6448b png_create_read_struct + 27
3   libwx_osx_cocoau_core-3.0.dylib	0x00000001088b925a wxPNGHandler::LoadFile(wxImage*, wxInputStream&, bool, int) + 106
4   libwx_osx_cocoau_core-3.0.dylib	0x00000001088a5bbb wxImage::DoLoad(wxImageHandler&, wxInputStream&, int) + 251
5   libwx_osx_cocoau_core-3.0.dylib	0x00000001088a6026 wxImage::LoadFile(wxInputStream&, wxBitmapType, int) + 182
6   libwx_osx_cocoau_core-3.0.dylib	0x00000001088a353a wxImage::LoadFile(wxString const&, wxBitmapType, int) + 154
7   libwx_osx_cocoau_core-3.0.dylib	0x0000000108762e15 wxBitmap::LoadFile(wxString const&, wxBitmapType) + 661
8   XS.bundle                     	0x00000001080ab4ca Slic3r::PresetCollection::load_bitmap_default(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 122
9   XS.bundle                     	0x00000001080bcffd Slic3r::PresetBundle::PresetBundle() + 1197
10  XS.bundle                     	0x0000000107e358f8 XS_Slic3r__GUI__PresetBundle_new(interpreter*, cv*) + 200
11  perl5.22.0                    	0x00000001079e4d53 Perl_pp_entersub + 2659
12  perl5.22.0                    	0x00000001079dc786 Perl_runops_standard + 22
13  perl5.22.0                    	0x000000010795307e Perl_call_sv + 1054
14  Wx.bundle                     	0x0000000108fc23e0 XS_Wx___App_Start(interpreter*, cv*) + 320
15  perl5.22.0                    	0x00000001079e4d53 Perl_pp_entersub + 2659
16  perl5.22.0                    	0x00000001079dc786 Perl_runops_standard + 22
17  perl5.22.0                    	0x00000001079528ce perl_run + 942
18  perl5.22.0                    	0x000000010792d21b main + 155
19  libdyld.dylib                 	0x00007fff936505ad start + 1

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff8cb7d5e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff94f73578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff94f71341 start_wqthread + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8cb7defa kevent_qos + 10
1   libdispatch.dylib             	0x00007fff9906b165 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff9906adcd _dispatch_mgr_thread + 52

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8cb7d5e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff94f73578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff94f71341 start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff8cb7d5e2 __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff94f73578 _pthread_wqthread + 1283
2   libsystem_pthread.dylib       	0x00007fff94f71341 start_wqthread + 13
[... 308 lines cut ...]
TOTAL                            484.4M      601 

Model: MacBookPro5,3, BootROM MBP53.00AC.B03, 2 processors, Intel Core 2 Duo, 2.8 GHz, 4 GB, SMC 1.48f2
Graphics: NVIDIA GeForce 9400M, NVIDIA GeForce 9400M, PCI, 256 MB
Graphics: NVIDIA GeForce 9600M GT, NVIDIA GeForce 9600M GT, PCIe, 512 MB
Memory Module: BANK 0/DIMM0, 2 GB, DDR3, 1067 MHz, 0x80CE, 0x4D34373142353637334548312D4346382020
Memory Module: BANK 1/DIMM0, 2 GB, DDR3, 1067 MHz, 0x80CE, 0x4D34373142353637334548312D4346382020
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x8D), Broadcom BCM43xx 1.0 (5.106.98.101.28)
Bluetooth: Version 4.4.6f2, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Samsung SSD 850 EVO 500GB, 500.11 GB
Serial ATA Device: MATSHITADVD-R   UJ-868
USB Device: USB 2.0 Bus
USB Device: Card Reader
USB Device: USB 2.0 Bus
USB Device: Built-in iSight
USB Device: USB Bus
USB Device: Apple Internal Keyboard / Trackpad
USB Device: IR Receiver
USB Device: USB Receiver
USB Device: USB Bus
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus:

alsand61, 2017-12-11:
Same problem for me, Slic3r 1.38.x not starting on a MacPro (late 2008) running OS X 10.11.6

But runs fine on an iMac (21.5-inch, Mid 2011) also running OS X 10.11.6

Tried removing ~/Library/Application\ Support/Slic3r/ but still no joy...

bubnikv, 2017-12-11:
Does Slic3r PE Slic3r 1.38.2 start at least? For 1.38.3 I recompiled the wxwidgets library.

I have a gut feeling, that the compiler that we used to compile Slic3r on, produced some CPU instruction, which is not supported by your outdated CPU. What exactly is your computer type and what CPU does it run on?

fiveangle, 2017-12-11:
Mac Pro "Quad Core" 2.8 (2008)2.8 GHz Q. Core Xeon E5462
Mac Pro "Eight Core" 2.8 (2008)2.8 GHz Q. Core Xeon E5462 x2
Mac Pro "Eight Core" 3.0 (2008)3.0 GHz Q. Core Xeon E5472 x2
Mac Pro "Eight Core" 3.2 (2008)3.2 GHz Q. Core Xeon X5482 x2

bubnikv, 2017-12-11:
I have reviewed the dmg archive contents of 1.37.xx and 1.38.xx and I see that 1.38.xx now bundles the libpng and other libraries, while it should not. I am in the process of finding how it was done before.

rambopierce, 2017-12-11:
Sounds like excellent progress. Thank you.

alsand61, 2017-12-11:
> Does Slic3r PE Slic3r 1.38.2 start at least? For 1.38.3 I recompiled the wxwidgets library.

No, neither 1.38.2 nor 1.38.3 will start.

> I have a gut feeling, that the compiler that we used to compile Slic3r on, produced some CPU instruction, which is not supported by your outdated CPU. What exactly is your computer type and what CPU does it run on?

As **fiveangle** correctly stated, mine is a Mac Pro "Eight Core" 2.8 (2008), running on 2 Quad Core Xeons.

Not exactly cutting edge technology, but still (mostly) works like a charm.

Thanks for looking into this, I too have a feeling that this is a compiler-related issue.

alsand61, 2017-12-12:
> This build should work on all OSX 64bit machines with OSX >=10.9
Slic3r-1.38.3.25-prusa3d-full-g74cb74f-201712121849.dmg.zip

YES !! This build runs just fine on all my machines.
Great job, once again...

> Please note that the Slic3r user profiles were moved from
~/Library/Application\ Support/Slic3r/
to
~/Library/Application\ Support/Slic3rPE
 also
 the print/filament/printer subdirs are now moved to
~/Library/Application\ Support/Slic3rPE/presets

No problem, I exported a config bundle from the previous version and loaded into this one.

bubnikv, 2017-12-13:
Thanks, closing this issue.

closed 2017-12-13 by bubnikv