prusa3d/PrusaSlicer #623: Minimum layer height is not 0.05 mm as expected

sepotvin, 2017-12-11:
In all the printer configurations, it looks like the minimum layer height as configured in "printer settings->Extruder n->Layer height limits->Min" is 0.07 mm by default instead of 0.05 as would be expected from the minimum layer resolution available to the MK2/MK2S/MK3. It forces the support to be mostly not synchronized with the printing layers with weird z values and makes Slic3r not respect the contact Z distance of the supports as it looks like it's not able to find a solution that fits all the constraints. With the Min height set to 0.05, I'm able to get the contact Z distance for supports down to 0.05 mm which is much better than the default most of the time (even if I would like to be able to set it to a small negative value to get more squish, but this is another issue that is already tracked on the forum).

bubnikv, 2017-12-11:
the 0.07 layer height was found the thinnest to work reliably on our i3 MK2 printers. The layer 0.05 did not print reliably with some materials.

You are likely correct that the support algorithm may have difficulty to find a set of layer heights to fulfill all the constraints. This is a price for the generality of the support algorithm. I admit there is certainly a room for improvement.

rtyr, 2020-03-23:
I believe we can close this issue. Reason for current value was explained.

closed 2020-03-23 by rtyr