prusa3d/PrusaSlicer #624: [Bug/Suggestion] Slic3r behavior if unknown placeholder variable is encountered in custom g-code

FHeilmann, 2017-12-11:
### Version
_Version of Slic3r used goes here_

Slic3r Prusa Edition 1.38.3-prusa3d-linux64

### Operating system type + version

Ubuntu 16.04 and Windows 10

### Behavior
* _Describe the problem_
    * If an unknown placeholder is encountered (e.g. typos or incomplete migration from S3D) the insertion of custom g-code is aborted entirely.
* _Steps needed to reproduce the problem_
    * Put a placeholder in the start-gcode that doesn't exist (e.g. [first_layer_humidity])
    * Slice the file
* _Expected Results_
    * The unknown placeholder evaluates to 0 or an empty string
* _Actual Results_
    * The insertion of the g-code snippet is aborted. Nothing is inserted in place of the invalid placeholder and all gcode after the unknown placeholder is lost.

_Is this a new feature request?_
Sorta. I guess aborting insertion of G-Code may be one way to deal with unknown placeholders, but in my opinion, simply not inserting anything or inserting 0 would be the preferred way to deal with unknown placeholders.

bubnikv, 2017-12-11:
The upcoming beta will have the error reporting much improved, so you will get an accurate information about the failing placeholders, and the G-code will be left on the hard drive with a .tmp suffix.
I think a good error reporting is better than failing silently.

bubnikv, 2017-12-11:
You may try the continuous integration build from here:
https://github.com/prusa3d/Slic3r/issues/619

Please note that we have moved the presets from Slic3r to Slic3rPE/presets, so you will need to copy your profiles.

bubnikv, 2017-12-14:
Error handling of incorrect macros in the custom G-code script has been improved in 1.38.4. Closing.

closed 2017-12-14 by bubnikv