prusa3d/PrusaSlicer #625: [Suggestion] Allow disabling of automatic temperature command insertion

FHeilmann, 2017-12-11:
### Version
_Version of Slic3r used goes here_

1.38.3

### Behavior
* The automatic temperature command insertion allows people to use Slic3r without having to fiddle with the start and end gcode scripts too much. Advanced users may wish to disable the inclusion of these commands. Either due to their specific setup or firmware considerations (e.g. RRF deprecated the M104 command)
* Please either include a toggle in the advanced section or maybe introduce some magic comment that will disable the automatic insertion of any heating commands.
_Is this a new feature request?_
Yes

bubnikv, 2017-12-11:
Are you aware, that you may override the temp G-codes in the custom start G-code?

FHeilmann, 2017-12-11:
Yes, but in my case RRF wants G10 to set the temperature with M104 being deprecated, so unless I also put a M104 in my start gcode, Slic3r will do it.

bubnikv, 2017-12-11:
What is RRF?
Shouldn't be that RRF be rather added as a new machine type?

FHeilmann, 2017-12-11:
Apologies!
I mean RepRapFirmware, specifically the DC42 fork used on the Duet boards. M104 has been deprecated there (though it is still supported) in favor of G10 R and S parameters.

bubnikv, 2017-12-11:
Is there any notion of "set temperature and wait" or "set temperature, don't wait" in the Duet firmware?

FHeilmann, 2017-12-11:
Theres G10 to set the active and standby (if another tool is selected) temperature and M116 to wait for all active tools to reach their target temperature. Bed is still handled by M140 and M190.

So my heatup sequence would looks like this:

```
G10 P0 R0 S[first_layer_temperature]
M140 S[first_layer_bed_temperature]

...
do some stuff that can be done in parallel to heating to save time
...

M116
M190 S[first_layer_bed_temperature]
```

Without any special measures taken Slic3r will automatically include the M104 in before the start gcode and another M109 after the start gcode.

Like I said this issue is not a severe one since I can work around it by including empty M104 and M109 commands in my start g-code. I consider it a cosmetic change. Since Slic3r already performs a check if the heating commands are included in the start code, I thought it might be easy to override the result of that detection process

closed 2020-10-23 by lukasmatena