prusa3d/PrusaSlicer #626: Cpouple of speed control issues?

theopensourcerer, 2017-12-11:
### Version
_Version of Slic3r used goes here_
1.37.2 and 1.38.3
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Ubuntu 17.10

### Behavior
Two issues related to speed... I mostly print PETG which likes it fairly slow...

I have 3 top and bottom layers and I noticed the other day whilst using 1.37.2 that the first of the top three layers printed extremely fast - I had to wath the print and pull off some strings before they caught on the next movement. The 2nd and third top layers were fine speed-wise.

Today I thought I'd try 1.38.3. Using the same profile, the print was way too fast for PETG - it wasn't sticking to the bed.

I went back to 1.37.2 and it is now printing at a speed I would expect.

I am not sure if this is really a bug or not but wanted to just make a comment and see if anyone else has seen anything similar recently.

Attached is a screenshot of my default PETG print speed settings tab. 
[image]

Sebastianv650, 2017-12-11:
The first solid layer over sparse infill is printed with bridging speed, and your setting is 50mm/s for this.

theopensourcerer, 2017-12-11:
@Sebastianv650 Thanks - That's helpful. What percentage is "sparse" up to? I was using 20% or 25% rectilinear.

Sebastianv650, 2017-12-12:
Everything <100%.

theopensourcerer, 2017-12-12:
@Sebastianv650 Really? That seems a bit weird? So I can't control the speed of the _first_ top layer without changing the bridging speed - hmm... Maybe this should be a feature request? 

But thanks - at least I can now workaround it for designs that do not need relatively large bridges.

Sebastianv650, 2017-12-12:
Just slice a cube and have a look ;-) http://www.gcodeanalyser.com/

It makes sense as the first layer over infill is some kind of a bridge. With cubic infill it's even not that small using normal percentages. If you need perfect looking, extreme brindging then you might need other speeds than with usual top fill, that's true.

theopensourcerer, 2017-12-12:
That's a cool tool - thanks!

bubnikv, 2017-12-14:
Is the issue clear now and can I close this issue?

theopensourcerer, 2017-12-14:
Kind of I guess. The fact that the first of the top layers prints at bridging speed seems a bit odd to me and I Cura doesn't do that. I'll close it but maybe add something to a wish-list or update the documentation so it is clearer?

closed 2017-12-14 by theopensourcerer

bubnikv, 2017-12-15:
The slic3r manual is maintained by Alessandro.
Please post your proposal here:
https://github.com/alexrj/Slic3r-Manual/issues

Thank you.