prusa3d/PrusaSlicer #628: [Suggestion/Request] include colorprint extruder change in Slic3rPE

mavu, 2017-12-12:
### Version
1.38.3

### Operating system type + version
Linux debian 

### Behavior

Now that variable layer height is enabled for multi-material printers, it would be really really nice if we could get color-print features built into Slic3r.

Currently (as far as I could find out) you cannot set a material change by layer in Slic3rPE, except by cutting the model, importing the same model again as modifier mesh, and assigning extruders.

Is it planned to include an easier option of doing material change on specific layers?

bubnikv, 2017-12-13:
We plan to implement that. The most We have a new hire @YuSanka to help out with the UI and this feature is mostly about the UI, the backend changes are quite simple.

mavu, 2017-12-13:
That is very good to hear, thanks :)

MStylesMS, 2018-01-07:
This would be a nice feature as I have several models that need this.  It would also be nice to be able to choose between A) user prompt to insert new filament and B) switch to a specific extruder that is already loaded.  B allows for fully automated printing but only up to 4 filaments.  A would allow more that 4 different colors but require user attention during the print process.

maccesch, 2018-06-23:
I'd love to see this too. It would be important though to be able to either put custom G-Code after the `M600` or at least to specify some extra extrusion before the print continues.

Right now I'm manually editing the G-Code because I always need 1mm extrusion right before the print resumes.

maccesch, 2018-06-25:
Also the `M600` should be inserted right after the retract and the deretract should be then shortened. The current implementation of ColorPrint inserts the `M600` after the deretract which of course then leaves some unwanted plastic

bubnikv, 2018-08-02:
Hopefully @vojtechkral will find time to implement it for Slic3r PE 1.42

alexiri, 2018-08-10:
I'm really looking forward to @vojtechkral's progress on this. I usually use the [javascript app](http://www.prusaprinters.org/color-print/) but it's terrible. I've had it cut the end of my gcode file and miss inserting color changes where it should. It's totally untrustworthy!

bubnikv, 2018-12-11:
The M600 aka "color print" was implemented in Slic3r PE 1.42 alpha to be released soon. Closing.

closed 2018-12-11 by bubnikv