prusa3d/PrusaSlicer #630: MK2 MMU Randomly stops printing

whoishack, 2017-12-13:
### Version
1.38.3

### Operating system type + version
Windows 10 pro

### Behavior
Prusa mk2 with MMU just started doing this today and random items. At first I thought it was the model but it's even doing it to previous sliced models that have printed fine.  Will be printing no problem but the extruder will randomly veer off course - instead of turning at a corner it would continue straight and extrude - then would return to position and keep printing, essentially making the model hairy. It would eventually just stop printing. All temps still up but it's almost like all steppers just stop. No jams, no messages, no shown issues at all. It's random when it happens. Sometimes 97% finished, sometimes 3%.
Uninstalled and reinstalled slic3r both 1.38.3 and the version on the prusa site. Flashed firmware again. Can slice the same model in normal mk2 mode and send it to my other prusa(using SD card on both) and it prints just fine. Not many files on the SD but thought maybe it was using too much mem so I cleared it but that didn't solve the issue. 

#### STL/Config (.ZIP) where problem occurs
Will upload an STL but config and stuff is all stock since reinstalling a fresh copy. 
[test.zip](https://github.com/prusa3d/Slic3r/files/1553756/test.zip)

bubnikv, 2017-12-13:
I started your G-code on one of our printers. Let's see.

> Will be printing no problem but the extruder will randomly veer off course - instead of turning at a corner it would continue straight and extrude 

Do you have a picture or a video?

Thanks.

bubnikv, 2017-12-13:
Hmm, it printed fine. I will try to flash the latest firmware and try again.

bubnikv, 2017-12-13:
I have updated the firmware to the current 3.1.0 and now I am starting the 2nd print.

Do you have your printer connected over USB to anything (PC or OctoPrint)?

whoishack, 2017-12-13:
I do have pics, I meant to post last night but got way too busy. 
I've attached a zip with numerous images. the last shows what it should look like right off the bed. 
[Photos (3).zip](https://github.com/prusa3d/Slic3r/files/1556198/Photos.3.zip)

bubnikv, 2017-12-13:
Didn't you send me the working G-code instead of the broken one by mistake?
The G-code seems to print fine even with the latest firmware.

whoishack, 2017-12-13:
Nope, that's the exact gcode i've been running when it goofs. It seriously makes no sense

whoishack, 2017-12-13:
I re-calibrated XYZ and it's not printing like it was as in it's actually finishing. It's still not printing good though or even remotely like it was. Everything is incredibly stringy and the walls are worse than they have ever been.. this is all with PLA that i've printed no problem before. 
[image]

whoishack, 2017-12-13:
Flashed to a previous firmware, then reflashed the latest off git. Running that test code now

bubnikv, 2017-12-14:
Isn't your issue caused by the your printer hardware? It may be the X or Y
pulley slipping at high accelerations and jerks. It may be, that some
Slic3r profiles are tuned for higher accelerations.

This isn't as uncommon as it sounds. It happened to me once on a first
printer I got at Prusa Research. The slipping pulleys are likely to be
detected by a firmware auto test.

whoishack, 2017-12-14:
I was never notified of this, i'm sorry. That may be, but I even turned all speed and jerk settings down and it didn't solve it. 
However, it seems down-flashing then reflashing the latest solved the issue. Been printing all day and haven't had an issue.

closed 2017-12-18 by whoishack