prusa3d/PrusaSlicer #632: Changes to settings do not show (Modified) in the drop down menu

scott-pett, 2017-12-14:
### Version
1.38.3-Prusa3D

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Mac OSX 10.11.6

### Behavior
* When changes have been made to Print Settings; Filament Settings etc the drop down menu used to add (Modified) to the name to warn you had to save the settings. This no longer occurs until AFTER you have attempted to change Filament or Print etc.

* Make changes to the Print Settings tab. Attempt to move to another print setting, a warning popup is generated to ask whether you wish to save or discard the changed. Click NO and then the drop down menu item name changes

 * _If this is a command-line slicing issue, include the options used_ - NA

* Drop Down name changes to include (Modified) when a change is made to the settings.

* Dropdown name doesn't change until after the warning popup.

  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_ - NO

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2017-12-14:
Please try with the just released 1.38.4. It should be working now.

freiser77, 2017-12-20:
@bubnikv: It's still not working. At least at my machine.

bubnikv, 2017-12-20:
Thanks, I see it now. It is an OSX / wxWidgets specific issue.

bubnikv, 2017-12-20:
This build should behave correctly.
[Slic3r-1.38.4.13-prusa3d-full-g4572fe6-201712201247.zip](https://github.com/prusa3d/Slic3r/files/1575594/Slic3r-1.38.4.13-prusa3d-full-g4572fe6-201712201247.zip)

bubnikv, 2018-02-12:
I suppose this has been fixed in 1.38.6. Closing.

closed 2018-02-12 by bubnikv