prusa3d/PrusaSlicer #633: Slicer generating dots next to mesh when using soluble interface

ir-fuel, 2017-12-14:
### Version
1.38.3

### Operating system type + version
macOS 10.12.6

### Behavior

When slicing with Soluble Support Interface enabled I get weird artifacts next to the mesh.
These are not there when slicing without supports

[image]
[image]
[image]

[config.txt](https://github.com/prusa3d/Slic3r/files/1560948/config.txt)

[all.stl.zip](https://github.com/prusa3d/Slic3r/files/1560949/all.stl.zip)

FidelCapo, 2018-02-14:
Thank you for report. Thats weird artifacts are supports from inside. Thats because of how Slic3r-PE works with models and supports and also the shape of your model.

bubnikv, 2021-02-24:
Leakage of the supports through object walls has been improved with fb7b9950501d01164955b7dd29f163005552234e
[image]

closed 2021-02-24 by bubnikv