prusa3d/PrusaSlicer #634: automatic insertion of g-code at the start of a file in kind of wrong order

RudolphRiedel, 2017-12-15:
### Version
1.38.4
### Operating system type + version
Win 10
### Behavior
Apart from the customizable G-code that can be setup under Printer Settings/Custom G-Code/Start G-code and Filament Settings/Custom G-code/Start G-code, Slic3r adds a few lines on it's own.

My g-code files look like this:
M107 ; disable fan
Printer Settings/Custom G-Code/Start G-Code
Filament Settings/Custom G-code/Start G-code
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0 ; reset extrusion distance

I propose to make it look like this:
M107 ; disable fan
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0 ; reset extrusion distance
Printer Settings/Custom G-Code/Start G-Code
Filament Settings/Custom G-code/Start G-code

Also, the very next lines kind of annoy me right now:
G1 Z0.280 F4800.000 ; move to next layer (0)
M117 printing layer 0
G1 E-6.50000 F4800.00000 ; retract
G92 E0 ; reset extrusion distance
G1 Z0.480 F4800.000 ; lift Z
G1 X128.466 Y134.044 F4800.000 ; move to first skirt point
G1 Z0.280 F4800.000 ; restore layer Z

The problem I have with this is that it sets down the nozzle and shoots across the build-plate.
And it messes up my nozzle priming start-code, the skirt is printing with the start of the third line, if at all.
Without skirt the first layer of the print would be missing some lines right now.

I rather have it this way:

G1 X128.466 Y134.044 F4800.000 ; move to first skirt point
G1 Z0.280 F4800.000 ; restore layer Z

Yes, that assumes that Z is up after the Start G-Code.
This can be assured by either parsing the users g-code and only adding lift when necessary instead of converting that lift to putting the nozzle down as it is now.
Or by adding a small relative Z movement before moving to the frist skirt point.

RudolphRiedel, 2017-12-15:
After 2 hours of playing some more with the startup g-code I have to admit that the proposition to move the G21... lines is no good.
The following code has to ensure that everything is done right so G21 and so on have to be issued after the user defined codes.
If you need it you can still add it to your custom g-code.

However that unwanted Z-lift or rather moving of Z down before moving to the start point is what made me tinker with the g-code to begin with and that one is still valid.

Like this it is totally fine for me:
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0 ; reset extrusion distance
;G1 Z0.280 F4800.000 ; move to next layer (0)
M117 printing layer 0
G1 E-6.50000 F4800.00000 ; retract
G92 E0 ; reset extrusion distance
;G1 Z0.480 F4800.000 ; lift Z
G1 X142.231 Y144.191 F4800.000 ; move to first skirt point
G1 Z0.280 F4800.000 ; restore layer Z
G1 E6.70000 F1200.00000 ; unretract
M204 S500 ; adjust acceleration

However to ensure that the Z axis is lifted, why not make it look like this:
G21 ; set units to millimeters
M82 ; use absolute distances for extrusion
G92 E0 ; reset extrusion distance
G91 ; set to relative positioning
G1 Z0.480 F4800.000 ; lift Z
G90 ; use absolute coordinates
M117 printing layer 0
G1 E-6.50000 F4800.00000 ; retract
G92 E0 ; reset extrusion distance
G1 X142.231 Y144.191 F4800.000 ; move to first skirt point
G1 Z0.280 F4800.000 ; move layer Z to starting height
G1 E6.70000 F1200.00000 ; unretract
M204 S500 ; adjust acceleration

I just tested this and it works flawlessly, this results in a little Z-lift after the user startup-code
instead of bringing Z down allmost to the bed.

As of why this annoys me, there is no way to change that behaviour other than editing the g-code files by hand.
I prime the nozzle by extruding some mm of filament with Z at 15 mm, so when Z moves down all the way again before moving to the starting point it hits that pile of filament when I did not remove it in time. When Z is maintained however that is not an issue because that pile is not moved along to the printing area.

bubnikv, 2017-12-20:
Frankly I think you are just over complicating things. It seems to me that your only real problem is, that Slic3r lowers the nozzle at the position where you extruded your thick line. A simple cure is to move the print head a bit from that position at the end of your start G-code. I don't really see any other showstopper right now for you.

RudolphRiedel, 2017-12-22:
To quote the only relevant sentence from my first comment:
"The problem I have with this is that it sets down the nozzle and shoots across the build-plate."

The g-code above is what Slic3r adds.
It puts down the nozzle to the height of the first layer, then lifts Z to .48 mm which makes the first Z-Move command completely useless. Then it moves X/Y to the start point and sets Z down again.

My suggestion is to simplify things a bit.
Not issue that first useless G1 command to lower Z.
Make that first G1 command to move the nozzle up by a relative amount
or set it a bit higher, maybe 1 mm.

closed 2026-09-29 by prusaslicer-bot[bot]