prusa3d/PrusaSlicer #635: Color Indicators Not Showing on Plater tab.

rambopierce, 2017-12-15:
### Version
Version 1.38.4-prusa-win64

### Operating system type + version
Windows 7 - 64-bit

### Behavior
Missing Extruder color indicators on Plater tab (Windows 7 problem only).  The Color indicator column on the right side of the Plater tab listing the Extruders usually contains a two-color flag for each extruder.  One color indicating the filament color and one indicating the filament brand.  I can select on the empty white space and pick a color, but the color is not displayed.  Resizing the window does not help.  Thanks.

#### STL/Config (.ZIP) where problem occurs
Config is the initial config included with this release of Slic3r PE.
[image]

rambopierce, 2017-12-16:
After some more research I determined that this regression first occurred in Version 1.38.2.  The color indicators show up properly on Windows 7 in Version 1.37.2.

bubnikv, 2017-12-18:
Hmm, we tested on three Windows 7 computers and it seem to work fine. What
exactly are you trying to do?

When you installed the 1.38.4, did you let the Slic3r 1.38.4 install the
bundled profiles, or did you copy your original profiles to the new
Slic3rPE profile directory?
Would you please zip and send me your content of the Slic3rPE profile
directory? Maybe there is something wrong store there from the older Slic3r
installs?

rambopierce, 2017-12-19:
I can remove the Slic3rPE profile under AppData\Roaming and let it reinstall and the problem persists. I think it may have something to do with wxWidgets. Is that library statically linked in to slic3r or is that a run-time environment library on my Windows instance.  Maybe I have an old version or something?  I also have the documented problem of when I go to the filament tab color selector it doesn't work until I resize the panel.  Thanks.

bubnikv, 2017-12-20:
The wxWidgets library is linked dynamically, but if you just installed or unpacked our installation, slic3r should load the bundled wxWidgets library.

> I also have the documented problem of when I go to the filament tab color selector it doesn't work until I resize the panel.

Sorry, I dont't understand. Would you please be more specific?

rambopierce, 2017-12-20:
Sorry, I wasn't very clear. Similar, but not the exact same as issue #334.  The color picker on the 'Printer Settings' tab - Preview box - Extruder Color,  is dead until I resize the Window:
[image]

After any (even minor) window resize, it becomes visible and works:
[image]

Don't know if this issue is related to the issue documented above about missing color indicators on the Plater tab as resizing that window doesn't make the indicators appear, but I thought it could be a hint as to maybe it's picking up an outdated WXwidgets library.  This is obviously a local problem (unfortunately on my primary Windows workstation); I have had many versions of slic3r installed here.  The last version where the color indicators show up is 1.37.2.

bubnikv, 2017-12-20:
Thanks. I remember #334, but I was not really able to reproduce it and
because there was a workaround, it did not have a priority. We have now
@YuSanka working on porting this part of the UI from Perl to C++, so while
she is on it, she will hopefully fix #334 as well.

Regarding your issue, I would try to find a Windows 7 computer first, on
which I can reproduce your issue.

bubnikv, 2018-08-02:
We hope it is working now. No response, closing.

closed 2018-08-02 by bubnikv