prusa3d/PrusaSlicer #636: Support Interface is Incomplete or Not the full width of the support structure

rambopierce, 2017-12-15:
### Version
Version 1.38.4-prusa-win64
### Operating system type + version
Windows 7 64-bit

### Behavior
This issue is applicable to printing supports with only the interfaces in PVA.
Slice the attached model and focus on the support generated for the lower four horizontal pipes; below and above each pipe.  The interface is generated fully so that the bottom of the pipe is not marred by PLA, However, on top of the pipe the interface does not begin early enough and is not as wide as the pipe, so the PLA support wall is printed on top of the pipe where you want PVA interface only. 

Part is in RED, Support in YELLOW, and Interface in BLUE.

Below the pipe, interface looks good and the pipe is laid down on (blue) PVA only.
[image]

With the pipe completed about half way, you can see that it does not touch (yellow) PLA as it should.
[image]

Above the pipe however, (yellow) PLA should not be printed directly on the part.  There is some (blue) PVA interface, but not sufficient.
[image]

Along with the STL model to slice, I will attach a few screen shots of the 3D preview.
[Cube.zip](https://github.com/prusa3d/Slic3r/files/1564092/Cube.zip)

Thanks.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

FidelCapo, 2018-02-14:
Thank you for report. I was look at it and if you set Interface layers to 20, supports will be generated correctly.

rambopierce, 2018-02-14:
That’s interesting and I will try it. I had increased the interface layers with no success, but don’t recall trying exactly 20. Is there something specific about 20?

Thanks.

bubnikv, 2018-02-14:
The issue here is that Slic3r calculates the supports by polygonal areas
first, then from these polygonal areas it deduces, which area should be
marked as soluble and which should be marked as the support base. The
problem here is, that the one or two support areas are made of so thin
polygons, that there is no extrusion line fitting the polygon, so it looks
like Slic3r missed the interface layers.

So as a workaround you may just increase the number of interface layers
until they appear where you miss them. In my opinion though you will not
miss them here anyway, as the PLA support is usually very easy to remove
from the top of an object and it does not leave any scars (though with MM
it may leave spots of different color).

rambopierce, 2018-02-14:
Thanks. I will try it when I get a chance. I did notice scarring with the PLA interface which is why I’m trying to get soluble to work correctly. I also have the issue I fairly certain is documented in another case where the top few layers of the PLA support needs to be solid so that the soluble material will lay down and stick to it. So tempting to use soluble material for the entire support, but that is expensive.

bubnikv, 2021-02-24:
Tested with current master (after 2.3.0 release).

It seems the support interfaces are laid correctly. The issue may actually be a duplicate of #3673, where the concentric support interface pattern is not being filled in the center. 

The current master allows one to switch the soluble support interface pattern from Concentric to Rectilinear, in that case the G-code seems perfectly fine to me.

[image]

Let's close this one and keep #3673 open.

closed 2021-02-24 by bubnikv