prusa3d/PrusaSlicer #637: Incorrect GCODE for bed temps

quasipedia, 2017-12-16:
### Version
1.38.3-prusa3D-linux64

### Operating system type + version
Fedora 27 64bit running gnome Wayland session

### Behavior
I have two printers that use the same filament, one have a heated bed, the other doesn't.  My settings are first layer 65°C, other layers 60°C.

There are two problems (actual result):

1) The temperature for the initial layer is never set in the GCODE, only the line `M140 S60` (other layers) appears further down.
2) The line `M140 S60` also appears in the GCODE generated for the printer without heated bed , making the printer stop and wait forever (so: making the print fail).

The expected result would obviously be to;
1) Have a `M140 S65` line before the first layer
2) Not having `M140` commands in the GCODE for the printer without heated bed

_Is this a new feature request?_  Unless the problem is between keyboard and chair (i.e.: I didn't spot an option to say whether the printer has/does not have a heated bed) point number 2 is (add a config option for heated bed in the printer configuration).

#### STL/Config (.ZIP) where problem occurs
[config.zip](https://github.com/prusa3d/Slic3r/files/1564349/config.zip)

bubnikv, 2017-12-16:
would you please retest issue 1 with slic3r pe 1.38.4? this issue should be
already fixed.

quasipedia, 2017-12-16:
Yes, I can confirm n.1 is fixed, thank you.  Will close this and open a feature request for n.2.

closed 2017-12-16 by quasipedia