prusa3d/PrusaSlicer #638: "Heated bed" should be a toggable option from printer configuration

quasipedia, 2017-12-16:
### Version
At least up to 1.38.4

### Operating system type + version
Any OS

### Behavior
Slic3r always emit the GCODE for setting the bed temperature, when slicing, regardless of the target printer having a heated bed or not.  On [some? all?] printers without a heated bed, this will make the build fail, as the printer will wait indefinitely for an event (the bed being hot) that will never occur).

**Slic3r should have a toggable checkbox ("Has heated bed") under the printer configuration**, and should omit entirely `M140` and `M190` instructions (or any other bed temperature related ones) from the GCODE output, when using printers without a heated bed.

quasipedia, 2017-12-17:
For those with my same problem, a workaround for now is to create a script with the following content:

    #! /usr/bin/env sh
    sed -i '/^M1[4|9]0 S.*$/d' $1

and make it run automatically at GCODE post-processing time via Print Settings → Output options → Post-processing scripts.  Remember to make it exectutable via `chmod +x <name of script>`.

bubnikv, 2017-12-20:
I know the upstream Slic3r has it. We will merge it into 1.39.xx.

closed 2026-09-29 by prusaslicer-bot[bot]