prusa3d/PrusaSlicer #639: Add notification of new available releases

quasipedia, 2017-12-16:
### Version
At least up to 1.38.4

### Operating system type + version
At least Linux (Appimages) possibly also other OS?

### Behavior
Due to lack of proper packages/repos linux users need install Slic3rPE via Appimages.  However - among their numerous drawbacks - Appimages do not provide a self-update mechanism out-of-the-box.

For mitigating at least this specific limitation, it would be very simple to at least offer a notification system, checking for the availability of new releases (to be then manually downloaded).

A dead-easy way to power this functionality is to use the GitHub API.  The relevant endpoint is [here](https://developer.github.com/v3/git/refs/). :)

quasipedia, 2017-12-16:
If anybody is interested in stop-gap alternative, I created a [simple script](https://gist.github.com/quasipedia/52a27c945b5e395f2e00abdb6ff30187) that can be used to automate the process.

bubnikv, 2017-12-20:
Thanks, this is something for the 1.39.xx release, I don't think it will make it into 1.38.xx

bubnikv, 2018-03-14:
@vojtechkral works on downloading the configuration profiles into Slic3r. I am sure he will also add a notification on new Slic3r version being available for download.

bubnikv, 2018-08-02:
Notification implemented for a long time.

closed 2018-08-02 by bubnikv