prusa3d/PrusaSlicer #641: volumetric speed "Hint-Text" ignores "First Layer Speed" and "Extrusion multiplier"

michakrause, 2017-12-18:
### Version
1.38.4

### Operating system type + version
Debian Linux 9

### Behavior
Filament Settings -> Advanced -> Max volumetric speed
"Hint-Text" ignores settings:
Print Settings -> Speed -> First Layer Speed
and 
Filament Settings -> Filament -> Extrusion multiplier

bubnikv, 2017-12-20:
I have fixed the dependency on the first layer speed settings https://github.com/prusa3d/Slic3r/commit/b2ba4ee34c5ccd7d8278939f33e51d1db3bfa9ef

Regarding the extrusion multiplier, it is really a question whether this multiplier shall be taken into account here. The question really is, what is the purpose of the extrusion multiplier: Is it used to underfill / overfill the print, or is it used to correct the flow rate to get a proper fill? I think in most of the cases it is the 2nd, so it should not be used inside the volumetric flow estimate formula.

I was actually thinking to provide both values: corrected with the extrusion multiplier and uncorrected as of now. But I think that would be too much.

michakrause, 2017-12-20:
Your right, I actually use the multiplier for both cases (printing flexibles with .5 to get a foam like part). But it's mostly for fine tuning flow rates.

closed 2017-12-20 by michakrause