prusa3d/PrusaSlicer #642: 1.38.4 M900 K0 being sent to output gcode at every "After Layer Change" In MM mode

tjcolejr, 2017-12-20:
### Version
1.38.4

### Operating system type + version
Win 10

### Behavior
In the Multi Material mode output gcode file the "filament gcode" is being canceled with a M900 K0 any time an "After Layer Change" is written(and possibly elsewhere).  M900 K200 is written at the beginning, and after a filament extruder change, but it is canceled by the next M900 K0.  This is preventing the proper implementation of linear advance.

Is this intentional?  If so I don't understand why.

#### STL/Config (.ZIP) where problem occurs
[K0.zip](https://github.com/prusa3d/Slic3r/files/1573741/K0.zip)

bubnikv, 2017-12-20:
This is on purpose and it is documented in the release log of 1.38.3:

   - Changed handling of filament_gcode_start and filament_gcode_end custom
   G-codes for a single extruder multiple material setup: At the start of a
   print, the filament_gcode_start is executed
   for the active extruder only, and the filament_gcode_start /
   filament_gcode_end are later executed at each tool change. When the Prusa
   MM wipe tower is active, the tool changes are handled a bit differently:
   M900 K0 is emited before the wipe tower extrusions start, and the
   filament_gcode_start code is executed after the wipe tower extrusions are
   finished. This rule effectively disables the linear advance over the wipe
   tower #568 <https://github.com/prusa3d/Slic3r/issues/568>.

You are supposed to set the linear advance value in the
filament_start_gcode. If you run the configuration wizard from the help
menu and you let it install the profiles, your filament profiles will be
updated with the correct linear advance value.

tjcolejr, 2017-12-20:
I have put the m900 k200 in the filament start gcode.  The k0 happens
before the cp blank start, but there is no k200 after the CP blank end.  It
is working correctly in the solid wipe tower routine, just not after the
hollow wipe tower routine.  I have sliced prepackaged settings and my own
settings and there is no difference. I was going to rewrite my issue with
specific examples, which is why I deleted my original.  I have spent hours
reading this gcode now and I get what you intend...switching off k during
wipe tower...it just isn't switching back on after the hollow wipe tower
routine.

bubnikv, 2017-12-20:
I see, thanks for the report. The issue is, I am applying the
filament_start_gcode only if there is an extruder change I suppose. I will
fix that.

bubnikv, 2017-12-20:
I have fixed that hopefully with https://github.com/prusa3d/Slic3r/commit/c8d14fb6178f4feea906f1509e9988fceceb0e26

After the change, the start_filament_gcode will be applied after each wipe tower operation. This is slowly getting pretty messy and the wipe tower is in a need of a redesign.

tjcolejr, 2017-12-20:
That's what it looked like to me.  I can just open the gcode in word pad
and do a find/replace on the blank end text to auto insert the k200.

I have tried to compile this myself and can't get past the perl gui line in
the install on Windows.  Tried everything, but I'm not a programmer so I
don't really know what I'm doing.  Im sure many dont even attempt it.  So
could we please get a new compiled release when this is fixed?  It is now a
pretty big deal since the filament k value is preinstalled in the filament
start gcode.
As for me.  I'm going to load Linux on a new drive partition and see if I
can work out my compiling ignorance.

bubnikv, 2017-12-20:
I will post a continuous build. I am thankful for any additional testing I
can get.

In regard to compilation on Windows, there may be some issues with the
described sequence and the provided precompiled packages. We will fix that
once this release is finalized.

bubnikv, 2017-12-20:
Please test with the continuous build attached. Thank you.

[Slic3r-1.38.4.13-prusa3d-win64-full-g4572fe6-201712201248.zip](https://github.com/prusa3d/Slic3r/files/1575723/Slic3r-1.38.4.13-prusa3d-win64-full-g4572fe6-201712201248.zip)

tjcolejr, 2017-12-20:
Thanks, I'll be happy to.  I'm at work right now but should be able to work
on it this evening.

tjcolejr, 2017-12-20:
Boom Fixed!  Now how do I sign up for these continuous builds!  That recent commit about the extrusion multiplier was the next potential bug I was investigating.

closed 2017-12-20 by tjcolejr

bubnikv, 2017-12-20:
Sorry, we do not publish the continuous builds. First, it only makes sense
to peek into the continuous builds during and after the beta stage. Second,
we did not have time to set up public access infrastructure to the results
of our build server. Maybe we will do that one day, but there is not enough
of us yet.

tjcolejr, 2017-12-20:
I understand totally.  I guess I have to get my compiling skills up to
speed!  Thanks for your hard work and responsiveness.

bubnikv, 2017-12-20:
I thank you for your vigilance. We are in a process of hiring software testers, but we don't have them yet, at least in the numbers we need. So any help with testing is very helpful. And the bug you found would take us some time to discover, so thanks again.