prusa3d/PrusaSlicer #643: Program crashes when [Suppress "- default -" presets:] is enabled

mloho, 2017-12-21:
### Version
1.38.4

### Operating system type + version
Windows 10 64-bit version 1703

### Behavior
Problem: Program (GUI) crashes when attempting to change any numerical settings value.
Steps:
- Open the program (GUI version)
- Navigate to preferences
- Enable [Suppress "- default -" presets:]
- Restart as per instructions
- Navigate to filament settings > Filament > Temperature > Bed > Other layers:
- Type a number into the box

Expected Results: The number gets updated with the newly entered one.
Actual Results: Program crashes.

bubnikv, 2017-12-21:
Sorry, I am not able to reproduce it. What is the active filament when it
crashes?
Would you please zip and send me the content of your Slic3rPE profile
directory?
It should be in "C:\Users\your_user_name\AppData\Roaming\Slic3r" or
"C:\Documents and Settings\your_user_name\Application Data\Slic3r"
Thanks.

>
> Version
>
> 1.38.4
>
> Operating system type + version
>
> Windows 10 64-bit version 1703
>
> Behavior
>
> Problem: Program (GUI) crashes when attempting to change any numerical
settings value.
> Steps:
>
> Open the program (GUI version)
> Navigate to preferences
> Enable [Suppress "- default -" presets:]
> Restart as per instructions
> Navigate to filament settings > Filament > Temperature > Bed > Other
layers:

bubnikv, 2017-12-21:
Sorry, I meant
It should be in "C:\Users\your_user_name\AppData\Roaming\Slic3rPE" or
"C:\Documents and Settings\your_user_name\Application Data\Slic3rPE"
the "Slic3r" directory has been renamed to Slic3rPE, don't send me the
Slic3r directory.
Thanks.

mloho, 2017-12-21:
Sorry, I flipped the option around. This issue can be seen when the option is disabled. I'll zip my directory just in case.

mloho, 2017-12-21:
[My Slic3rPE](https://github.com/prusa3d/Slic3r/files/1579031/Slic3rPE.zip)

bubnikv, 2017-12-21:
> Sorry, I flipped the option around

Now it makes sense to me :-)
I will retest. Thanks.

bubnikv, 2017-12-21:
Fixed with https://github.com/prusa3d/Slic3r/issues/643
Thanks for the report.

closed 2017-12-21 by bubnikv