prusa3d/PrusaSlicer #644: Default End G-code should turn off bed temperature as well

mloho, 2017-12-21:
It should have both:
```
M104 S0
M140 S0
```

bubnikv, 2018-08-02:
Anyone else thinks the same?

Do you want us to add these two lines into the default content of the end G-code, so one could remove the two lines, or do you want these two lines to be generated by default without showing them in the end G-code field?

Frankly I would rather not change anything, I think it works just fine as it is.

mloho, 2018-08-15:
At the time, I was thinking of safety concerns. If the printer that I use utilizes a heated bed, then wouldn't it be unsafe for this slicer to not turn off the bed at the end automatically? ...given that I had set the bed temperature to a value > 0 when I first setup my printer settings but I forgot to customize my end gcode.

rtyr, 2020-03-23:
I believe that mloho meant default content of the end g-code. It could be related to this https://github.com/prusa3d/PrusaSlicer/issues/638.

closed 2026-09-29 by prusaslicer-bot[bot]