prusa3d/PrusaSlicer #646: Complete individual objects causes the wipe tower to disappear

kursatu, 2017-12-23:
### Version
_Version of Slic3r used goes here_
1.37.2
_Use `About->About Slic3r` for release versions_
1.37.2 prusa3d win64
_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10, 64 bit
### Behavior
* _Describe the problem_

When you check the complete individual object in the output options, the following model 
encounters problems. The problems are visible when you export and open the gcode in pronterface or save it to the sd card.
* _Steps needed to reproduce the problem_
1- Slice the attached stl model using 0.15 mm soluble full setting.
2- Verify that it slices correctly by exporting gcode and opening it in pronterface.
3- Change the output options/complete individual object to true.
4- Slice and see that the wipe tower and initial color checks are there.
5- Export and open in pronterface. The wipe tower and the initial color checks are gone.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
[image]

  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
[key.zip](https://github.com/prusa3d/Slic3r/files/1583833/key.zip)

_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2017-12-24:
Thanks for the report.

I hoped there was a test implemented to disable the "complete individual
objects" choice when the multi-material print is active. So it is a glitch,
but it is not a show stopper.

rtyr, 2020-03-22:
Fixed in 2.2.0. "Complete individual objects" cannot be enabled for MM prints. Closing.

closed 2020-03-22 by rtyr