prusa3d/PrusaSlicer #647: M900 wit if statement in GCode, possibly not evaluating properly

Tirpitz93, 2017-12-23:
### Version
version_1.38.5

### Operating system type + version
Linux mint  bug
Windows ok

### Behavior
'M900 K{if printer_notes=~/.*PRINTER_HAS_BOWDEN.*/}200{else}30{endif}; Filament gcode'
in the gcode causes printer to crash.
doesnt occur on windows, they use the same config directory.

#### STL/Config (.ZIP) where problem occurs
[m900.zip](https://github.com/prusa3d/Slic3r/files/1584126/m900.zip)

bubnikv, 2017-12-24:
What Slic3r build did you run? Our AppImage, or our tar.gz? Or did you compile Slic3r yourself?

I tested our AppImage on Ubuntu 16.06.1 and I don't see any issue with the generated G-code.

What printer did you generate your G-code for? A single material, or a multi-material printer? Single color or multi color print? Did you update the Slic3r presets?

By the way, the zip you attached is empty. If you still have the issue, would you please upload your stl (or AMF for multi-color prins), your config and your G-code?

How does your printer crash?

Tirpitz93, 2017-12-24:
Slic3r build from source.
Single material MK2S
I had an issue with the support extruder, but setting it from 0 to 1 fixed the issue where I could export configs. 
[M900.zip](https://github.com/prusa3d/Slic3r/files/1584964/M900.zip)
printer freezes at the end of the purge line.

bubnikv, 2017-12-25:
> Slic3r build from source.

Does our AppImage behave better?

> I had an issue with the support extruder, but setting it from 0 to 1 fixed the issue where I could export configs.

Would you please be more specific? Does that mean that the Windows build worked fine with the extruder 0, but the Linux build did not?

Isn't it the same issue as https://mail.google.com/mail/u/0/#inbox/1602ad43f57a8431 ?

bubnikv, 2017-12-25:
You generated the G-code using the upstream Slic3r.
# generated by Slic3r 1.3.0-dev on 2017-12-23 13:10:45

You need to build the Slic3r Prusa Edition to use the conditional syntax we have implemented in the Slic3r 1.38.xx series. This conditional syntax is not compatible with the upstream conditional syntax.

bubnikv, 2018-02-02:
Closing, this issue has been caused by incompatible G-code processors of the Slic3r PE vs. upstream Slic3r.

closed 2018-02-02 by bubnikv