prusa3d/PrusaSlicer #648: Wrong temperatures generated in gcode

ir-fuel, 2017-12-24:
### Version
1.38.3-prusa3d

### Operating system type + version
macOS Sierra

### Behavior
* Selecting PETG on extruder 4, Gcode generates PLA temperature settings in vase mode.

* Select generic PETG on extruder 4, set your only object in the scene to use extruder 4. Use vase mode. Slice and export GCode file
* Should have a hotend temp of 230 and a bed temp of 80
* Hotend temp op 210 and bed temp of 60

[image]
[image]
[image]

[tree_facets.stl.zip](https://github.com/prusa3d/Slic3r/files/1584770/tree_facets.stl.zip)

[tree_facets.gcode.zip](https://github.com/prusa3d/Slic3r/files/1584771/tree_facets.gcode.zip)

bubnikv, 2017-12-24:
This bug should be fixed in the current 1.38.5 release candidate. Please retest with 1.38.5 and if it works correctly, please close this issue. Thank you.

ir-fuel, 2017-12-24:
Indeed

closed 2017-12-24 by ir-fuel