prusa3d/PrusaSlicer #649: Allow changing the filename/title when sending to Octoprint

tuxmobil, 2017-12-24:
### Version

1.37.2-prusa3d-win64

### Operating system type + version

Win10 64

### Behavior

**Feature Request:** 

When sending a job to Octoprint, it would be nice to have a dialog (similar to saving a gcode file) to allow us to change the filename/title of the job that is sent to octoprint.

The current method of deriving the job title from the parts added to the print plate is often misleading (or colliding with existing jobs) if multiple parts are added to the print plate.

rongith, 2018-01-04:
Nice to have indeed but I think as there is long todo list, it would be treated lately.
I use [input_filename_base]-(P)[print_preset]-(F)[filament_preset]-(P)[printer_preset]-[version]-[timestamp].gcode to avoid colliding.

vojtechkral, 2018-03-15:
Seems to me this is a duplicate of #87

closed 2018-03-15 by vojtechkral