prusa3d/PrusaSlicer #650: Feature request: support for basic auth for octoprint

Tirpitz93, 2017-12-24:
### Version
future
### Operating system type + version
ALL

### Behavior
Enable connecting to octoprint behind a basic auth login.

AFAIK this is a new feature request.

vojtechkral, 2018-03-15:
Can you please elaborate what you mean by:

> Enable connecting to octoprint behind a basic auth login.

The standard way right now of authenticating to Octoprint is via the API key. Is this insufficient for you?

Tirpitz93, 2018-03-15:
My octo server is behind ha proxy with basic auth, so I cant upload directly. the whole server is inaccessible until authentication has succeeded.

vojtechkral, 2018-03-15:
@Tirpitz93 Ah, I see now what you mean. I'm not sure I see the reason for having this feature though. Can you describe your usecase or why you need it? Ie. specifically what the benefit is over authorization via an API key? Honestly I don't think we have enough cycles to implement it unless there is a convincing reason to do so...

vojtechkral, 2018-04-24:
No response for a long time, closing. Feel free to comment further if needed.

Tirpitz93, 2018-04-25:
because my printer is behind basic authentication before it even gets to the octoprint api authentication. AFAIK the only thing that would be required is sending authentication headers in the requests already being made.

Tirpitz93, 2018-05-01:
This has nothing to do with octoprint. In my setup basic authentication is managed through HAProxy, then I log in to octoprint again. which I manage through SSH. This means that octoprint continues to serve on plain http on the loopback IP and only the "secure" https haproxy faces the internet.

vojtechkral, 2018-05-01:
@Tirpitz93 Yes, I understand. Maybe the best solution would be the ability to add arbitrary http headers.

tracernz, 2018-05-01:
That's a rather peculiar setup. What I have done is install https://github.com/OctoPrint/OctoPrint-ForceLogin. So users must login to access anything through the web interface, but the API key still works for Slic3r etc. It does feel rather fragile relying on a plugin to load to enable auth requirement, and the default Octoprint setup is quite woeful in a number of ways in terms of security, compliance with FSH standard etc. sadly.

vojtechkral, 2018-05-02:
Well, perhaps the proxy serves other purposes as well. Although I still think even then it would be better to have the proxy verify the API key rather than using the basic auth. I'm not very keen on an explicit support for basic auth, since it would be interface-wise confusing with Octoprint login, but we could add support for it via arbitrary http headers, which might be useful for other reasons too. Still a bit low-prio at the moment though. But I'll keep it in mind.

Tirpitz93, 2018-05-02:
This seams to be a relatively common set up  AFAIK,
it is documented here: https://printoid.net/access-octoprint-from-the-internet/  as well as in other places.
This one odes mention the authkey in the config, but I haven't had a chance to test that set up.
https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian

Tirpitz93, 2018-05-02:
I agree that https should be used. and in my case is. but that guide had a helpful diagram to illustrate the setup.

alhazar313, 2018-05-26:
I too have haproxy enabled using that guide but configured with https and would love this feature

tracernz, 2018-05-26:
Did you consider the Octoprint-ForceLogin plugin first? Then the API key can be used without login (as is normal e.g. for GitHub API). Maybe there are some issues with this plugin I'm unware of, and if so I'd like to hear about them as I'm currently using it.

alhazar313, 2018-05-26:
I'd rather not have my security based on a plugin functioning, more secure using haproxy and https in my opinion.

vojtechkral, 2018-05-28:
Well this is emabrassing - it turns out we proably already suppor HTTP Baisc Auth (through libcurl). If you guys want to try it out, just use an URL in this shape:

    https://username:password@your-octopi-address.example.com/

I can't test it right now but if you can, it would be great if you could report whether it worked. Thanks!

alhazar313, 2018-05-28:
I'll give it a shot this afternoon :D

tracernz, 2018-05-28:
Ha! Good catch. :grin: 

> I'd rather not have my security based on a plugin functioning, more secure using haproxy and https in my opinion.

Yes, I am uneasy about that. I get the feeling that upstream OctoPrint are probably not that interested in patches to make it secure by default. :disappointed: I'm not sure if it's OctoPrint or OctoPi responsible but I'd also rather not have it spray log files in my home dir etc. too.

mcristina422, 2018-07-25:
@vojtechkral 
I was able to verify this worked in 1.40.1 against a nginx reverse proxy with basic auth

vojtechkral, 2018-07-26:
@mcristina422 Great!

I'll still leave this bug open as a reminder to better document the feature / make it discoverable, because the way it is right now it's pretty much invisible...

GreatGrizzly, 2018-07-31:
Having two extra fields in the Octoprint Upload section labeled Username and Password would be a great, easy edition to give this feature more exposure.

Tirpitz93, 2018-08-01:
@GreatGrizzly that is what I was originally hoping for, but @vojtechkral does have a point with it being a point of confusion for new users.
possibly if one was labeled `Server login` and one `Octoprint login`?

Tirpitz93, 2018-08-01:
@mcristina422 I tested it and got it to work this morning too. 
would be nice to have a pair of fields instead, and then use a credential store to encrypt the login details.

Sneakers82, 2018-12-10:
I've got Octoprint Version 1.3.9
HAPROXY setup using https on port 443. [Configuring-HAProxy](https://github.com/mastertork/Secure-OctoPi/wiki/Configuring-HAProxy)
Slic3r PE 1.41.2+Win64
I just tried the suggested `https://user:pass@1.2.3.4/`

I get the following Error Message:
> Could not connect to Octoprint: Peer certificate cannot be authenticated with given CA certificates (60)

I've double checked my Username, Password and API Key.

I also tried white listing my remote IP address which I thought would be a better solution. My browser will load Octoprint without the additional auth step, but Slic3r still gives me the same error. 
I'm assuming the trouble must be with the unsigned SSL?

vojtechkral, 2018-12-11:
@Sneakers82 

> I'm assuming the trouble must be with the unsigned SSL?

Yes. You'll need to import your self-signed certificate, this depends on the OS. On Windows, you'll need to import it into system certificate storage. On Linux / OS X there's a configuration line in Slic3r named `HTTPS CA file` where the cert file may be configured.

Sneakers82, 2018-12-11:
@vojtechkral
Thanks! I've added the self-signed certificate to the Trusted Root CA in windows.  I used this [guide]
(https://blogs.technet.microsoft.com/sbs/2008/05/08/installing-a-self-signed-certificate-as-a-trusted-root-ca-in-windows-vista/).
Octoprint was updated to 1.3.10

I get a new error message: SSL Connect Error (35)
Uninstalling the Cert I go back to Error (60)
I'm not sure about this one. Searching suggests curl is out of date but I doubt that.  Is there a log I can view that might help me out?

vojtechkral, 2018-12-11:
@Sneakers82 I'm afraid there's no way to get more detailed error message out of Slic3r at the moment since I'm not reading curl's verbose error buffer. (Which I very much should do and I'm putting it on my to-do list, it will be improved in the next release.)

In the meantime, I you can download `curl` for windows and try to use it from the command line in the same way as Slic3r in order to diagnose the problem. Here's a command line that does the same thing as Slic3r's `Test` button:

    curl -H 'X-Api-Key: <API-key>' https://<octoprint-host>/api/version

One thing that comes to mind is that by default curl [verifies that the hostname matches the name in the certificate](https://curl.haxx.se/libcurl/c/CURLOPT_SSL_VERIFYHOST.html) and we honor this default for security reasons. This means that if you're connecting to the OctoPrint via an IP address and the certificate instead specifies a hostname, it will be rejected.

Sneakers82, 2018-12-11:
@vojtechkral Thanks again! The hostname doesn't match the name on the cert. I'll give that a try tonight. 

I was wondering can I edit slic3r to use curl -k insecure?

vojtechkral, 2018-12-12:
@Sneakers82 no problem, glad you got it working.
Thanks for the links, that might be useful for other users...

jmaeso, 2020-04-29:
Hi, could be that special characters are not escaped? at least in the password part:

[image]
[image]

I would prefer not to change my ha-proxy password in the octoprint settings 😅
Am I missing something?

bubnikv, 2020-12-03:
basic auth is supported through the
```
https://username:password@your-octopi-address.example.com/
```
convention.

The following hint has been added to the tooltip
                  "Print host behind HAProxy with basic auth enabled can be accessed by putting the user name and password into the URL "
                   "in the following format: https://username:password@your-octopi-address/");
with 0a33aedc3adaa35886847504c607b843aae91d85

Raw URL was passed to libcurl, while it should have been URL encoded.
This was fixed with 1b720db

The changes will be in PrusaSlicer 2.3.0-beta2. Closing.

closed 2020-12-03 by bubnikv

bubnikv, 2020-12-03:
The commit 1b720db was not correct, I had to revert it.
I was calling the escape function over the complete http:/xxx/yyy line, which is incorrect. Certainly the http:// prefix must not be escaped.

I believe we have either to leave the escaping of special characters to you
https://www.werockyourweb.com/url-escape-characters/

or we have to add new input fields for user / password, escape them and combine them into an URL. That is a lot more work though and I don't think that the user base is large enough to justify the effort.

vojtechkral, 2020-12-04:
@bubnikv In newer curl versions (>= 7.62) there's `curl_url_get()` and `curl_url_set()` with the `CURLUPART_USER` and `CURLUPART_PASSWORD` flags [[1]](https://curl.se/libcurl/c/curl_url_get.html)
but I'm afraid there's again going to be the problem that some linux distros have curl older than that :-(

bubnikv, 2020-12-04:
@vojtechkral 
> In newer curl versions (>= 7.62) there's curl_url_get() and curl_url_set() with the CURLUPART_USER and CURLUPART_PASSWORD flags [1]

Nice, thanks. Now somebody "just" needs to implement that.

Frankly it would be better, if Octoprint implements HTTP digest authentization, which is much safer than the plain authentization.
https://github.com/OctoPrint/OctoPrint/issues/3854