prusa3d/PrusaSlicer #651: Feature request: Allow different brim settings for different objects

GurliGebis, 2017-12-26:
When printing different objects, it is sometimes an good idea to have a brim to help ensure it stays connected to the bed.

But when printing several different objects at the same time, it would make sense to be able to define the brim differently for the different types of objects.

apphead, 2018-07-21:
duplicate of https://github.com/prusa3d/Slic3r/issues/1033

I suggest closing this issue

apphead, 2018-07-21:
Nope, the other issue is more detailed ..

gmccauley, 2019-07-03:
+1 for this feature

rtyr, 2020-03-23:
Same as https://github.com/prusa3d/PrusaSlicer/issues/1033, https://github.com/prusa3d/PrusaSlicer/issues/2182, https://github.com/prusa3d/PrusaSlicer/issues/286 etc.. Closing this one.

closed 2020-03-23 by rtyr