prusa3d/PrusaSlicer #653: "Remember to check for updates at ..." URL is incorrect

jcwren, 2017-12-27:
### Version
1.37.2

### Operating system type + version
MacOS 10.13.2 

### Behavior
URL at bottom of screen that indicates to check for update points to http://github.com/prusa, instead of https://github.com/prusa3d

[image]

Patronics, 2018-01-11:
I encountered this as well (actually downloaded slic3r PE just to test this and contribute (my mk3 isn't due to arrive for a while)). I found this is based on the width of the slic3r window, and increasing the width of the window will reveal the full message and correct URL. The info bar always seems to go until a constant distance (arround 310 pixels from the right edge, at least on my system) from the right side of the screen. The whole message is visible from a window width of around 860 pixels or more. 
~~The relevant file to fix this might be Slic3r/lib/Slic3r/GUI/MainFrame.pm (that's where the text itself is defined, and some setup for the statusbar), but I don't know for sure or have enough experience to fix it myself.~~
Upon further examination, it seems the location where the status stops is where the cancel button and progress bar appear when slicing. I don't have the experience to fix this problem myself, but hopefully this information helps someone else narrow down and fix it.

TL;DR: The problem isn't the URL listed, but the width in which it can display. The statusbar it displays in blocks the rightmost 310 pixels of the message from view, where the cancel button appears.

bubnikv, 2018-01-11:
Thanks for heads up. We now have @YuSanka working on the Slic3r UI, she will look into it one day.

lukasmatena, 2019-08-26:
Still present in PrusaSlicer2.1.0-beta2. It appears to be OSX specific. Ellipsis ("...") is added on Linux and Win when the text is trimmed.

rtyr, 2021-01-20:
Already fixed. Closing.

closed 2021-01-20 by rtyr