prusa3d/PrusaSlicer #654: Segmentation fault (SIGFPE) when choosing 250000 baudrate on some printers

indrora, 2017-12-28:
### Version
1.38.5-prusa3d-linux64

### Operating system type + version
Linux (Ubuntu Xenial), AppImage build

### Behavior
* When I choose 250000 as the baud rate for my printer, it crashes with a SIGFLT floating point error
* Steps to reproduce:
  - Set baud rate to 250000
  - Press "test"
* _Expected Results_
  - everything is OK/I get a connection error
* _Actual Results_
  - SIGFPE crash. 

_Is this a new feature request?_ no

### strace output

```
indrora@chama /home/indrora/work/3dprinting > strace /home/indrora/bin/slicer-1.38.5-prusa3d.AppImage
[trim ... trim ... trim]
timerfd_create(CLOCK_MONOTONIC, TFD_CLOEXEC) = 13
epoll_ctl(12, EPOLL_CTL_ADD, 11, {EPOLLIN|EPOLLERR|EPOLLET, {u32=85365448, u64=85365448}}) = 0
write(11, "\1\0\0\0\0\0\0\0", 8)        = 8
epoll_ctl(12, EPOLL_CTL_ADD, 13, {EPOLLIN|EPOLLERR, {u32=85365460, u64=85365460}}) = 0
open("/dev/ttyACM0", O_RDWR|O_NOCTTY|O_NONBLOCK) = 14
fcntl(14, F_GETFL)                      = 0x8802 (flags O_RDWR|O_NONBLOCK|O_LARGEFILE)
fcntl(14, F_SETFL, O_RDWR|O_NONBLOCK|O_LARGEFILE) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, SNDCTL_TMR_START or TCSETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
epoll_ctl(12, EPOLL_CTL_ADD, 14, {EPOLLIN|EPOLLPRI|EPOLLERR|EPOLLHUP|EPOLLET, {u32=85360272, u64=85360272}}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, SNDCTL_TMR_START or TCSETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, SNDCTL_TMR_START or TCSETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, SNDCTL_TMR_START or TCSETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, SNDCTL_TMR_START or TCSETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCFLSH, TCIFLUSH)             = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, SNDCTL_TMR_START or TCSETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TCGETS, {B38400 -opost -isig -icanon -echo ...}) = 0
ioctl(14, TIOCGSERIAL, 0x7ffe892e2f50)  = 0
--- SIGFPE {si_signo=SIGFPE, si_code=FPE_INTDIV, si_addr=0x7f8f0c9e3f3f} ---
+++ killed by SIGFPE (core dumped) +++
Floating point exception (core dumped)
```

indrora, 2018-02-10:
Still happening on 1.39-alpha.

bubnikv, 2018-02-12:
@VojtechKral will look into this issue and he will provide a continues integration build to test on.

closed 2018-02-12 by bubnikv