prusa3d/PrusaSlicer #655: Bundled sample objects for MacOS version not updated in 2.1.0

triplepoint, 2017-12-28:
### Version
Release bundle version `2.1.0`. Slic3r PE version `1.38.5` (though the Slic3r version is irrelevant to this issue)

### Operating system type + version
MacOS, High Sierra

### Behavior
With a fresh install of the "drivers" bundle version `2.1.0`, the bundled gcode in the `Objects/MK2` package directory all have a May 23rd 2017 as their last-edit date, and a gcode header of:
```
; generated by Slic3r 1.30.0-prusa3d-release_candidate_1_3 on 2016-05-18 at 20:13:44

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
```

The other `Objects/` directories have similarly dated gcode files.

The release notes for 2.1.0 say:
```
- Updated MK2 sample objects with LA to print faster
- MK3 sample objects also updated
- New sample objects are copied to folder with drivers version appendix – no need to backup anymore
```

So it sounds like the new sample objects and organizational structure were missed in the MacOS release.  I didn't check the Windows release.

FidelCapo, 2018-02-13:
Try download last bundle 2.1.3 for Mac. I see there is Linear advanced gcodes.

closed 2018-02-13 by FidelCapo

bubnikv, 2018-02-13:
> So it sounds like the new sample objects and organizational structure were missed in the MacOS release. I didn't check the Windows release.

@FidelCapo went through the Windows vs. OSX bundled G-codes and verified, that they are identical for the current release of the Windows & OSX drivers.