prusa3d/PrusaSlicer #656: Vase bottom solid layers not following perimeters

m0n5t3r, 2017-12-29:
### Version
Slic3r Prusa Edition
Version: 1.38.3-prusa3d-linux64
Build: Slic3r-1.38.3-prusa3d-linux64-full-201712012144

### Operating system type + version
Ubuntu 16.04

### Behavior
* _Problem_: vases with slanted walls tend to leak due to most of the bottom layers not fusing with the perimeter; this seems to be due to the solid infill following the first layer's contour, instead doing the right thing and extending to contact the external perimeter of the current layer
* _Steps needed to reproduce the problem_: slice the attached model with the attached config (vase mode,  0.3mm layers, 5 bottom layers, 0.8mm external perimeter, 0.6mm default extrusion, "avoid crossing perimeters" and "ensure vertical wall thickness" disabled, other settings probably unimportant
* _Expected Results_: bottom solid layers contact the perimeter
* _Actual Results_: the solid infill follows the contour of the first layer:
[image]
 
#### STL/Config (.ZIP) where problem occurs
[vase-issue.zip](https://github.com/prusa3d/Slic3r/files/1592949/vase-issue.zip)

supermerill, 2018-01-03:
duplicate of #253 ?

m0n5t3r, 2018-01-03:
Meh, my search-fu is less; it is a duplicate, indeed. I tend to keep "ensure vertical shell thickness" disabled because I've had it remove all infill from areas that were supposed to be solid; gave it a try now, and with vases it does the opposite:
[image]

bubnikv, 2018-03-14:
Closing this one as it is a duplicate of https://github.com/prusa3d/Slic3r/issues/253 .

closed 2018-03-14 by bubnikv