prusa3d/PrusaSlicer #657: User modification/settings missing after upgrading Drivers

3d-gussner, 2017-12-30:
I like the idea of having Slic3r settings versions, but after updating/upgrading the drivers modifications and new filament settings aren't available anymore due to the new settings folder being used.

It would be great
1. To **notify** the user that user modifications are gone with the installation and may need manual adjustment.
2. I know that this is ambitious but a merge wizard would be great.

My workaround is to use winmerge to see the differences and manually modifying settings to my needs.
Dialing in Filament temps and M900 K values, print and print settings takes a while so it is a pity that these are "gone" after the upgrade.

I know that most people will be fine with the default print settings.

bubnikv, 2018-02-02:
Our new hire since yesterday @vojtechkral just started to work on automatic downloads of profile updates. The automatic update will notify you, if you modified the "system" profiles, and it will offer you to create backups of your modified profiles.

A merge tool would be cool, but it is a much more complex topic, though we may look into it one day.

vojtechkral, 2018-05-28:
With the introductiong of _System presets_ in Slic3r PE 1.40, this should be solved.

closed 2018-05-28 by vojtechkral