prusa3d/PrusaSlicer #660: I can't move an object out of the center of the bed. After dragging to a different corner, it auto-snaps back.

alchemistTi, 2018-01-01:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-01-01:
Open the preferences dialog (menu File->Preferences) and disable auto
centering.

bubnikv, 2018-02-02:
It was a user settings. Closing.

closed 2018-02-02 by bubnikv