prusa3d/PrusaSlicer #661: Missing parts of model in preview and when printing

ljohnsonnsb, 2018-01-02:
### Version
_Version of Slic3r used goes here_
1.37.2-prusia3d-win64
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Version 1703 
### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_
[Slic3r issue.zip](https://github.com/prusa3d/Slic3r/files/1597510/Slic3r.issue.zip)

Inner section of model is not printing or showing in preview mode but is showing in 3d mode.

_Is this a new feature request?_
NO
#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-02-02:
Slic3r does not handle intersecting bodies correctly. Slic3r will process your model correctly, if you fuse the intersecting bodies using boolean union. You may fix your STL with https://service.netfabb.com/login.php

bubnikv, 2018-05-10:
@FidelCapo Would you please test this object with the current 1.40 alpha and if it works, would you please close this issue and add the STL to the set of our test objects? Thanks.

FidelCapo, 2018-05-11:
Tested and fixed in Slic3rPE-1.40.0-alpha that will be released soon. Closing

closed 2018-05-11 by FidelCapo