prusa3d/PrusaSlicer #663: Printing in air at end of print

ir-fuel, 2018-01-03:
### Version
1.38.5

### Operating system type + version
macOS Sierra

### Behavior

At the end of the print, the print head moves to the front right of the bed, then goes to the front left of the bed, prints a bit in mid air and then stops.

Here you can see what happens:

[https://vimeo.com/249536566/0e098dcd8c](url)

#### STL/Config (.ZIP) where problem occurs
[Archive.zip](https://github.com/prusa3d/Slic3r/files/1600972/Archive.zip)

bubnikv, 2018-01-03:
I see you printed a single material print with a multi-material printed.
What you see is intentional, it simulates the final phase of re-shaping the
tip of the filament before it is parked into its cooling tube inside the Y
splitter. The printer has to eject the melted material from the nozzle
before it can park the filament.

ir-fuel, 2018-01-03:
I've never seen this before :) 
But ok, if it's intentional, all the better then

Thanks for the quick reply!

closed 2018-01-03 by ir-fuel

bubnikv, 2018-01-03:
It has always been there if you selected the "Original Prusa i3 MK2 MM
Single Mode" printer profile.

On the other side, the printer profile "Original Prusa i3 MK2
MultiMaterial" did not have this "filament parking" sequence built in,
which lead to jams after the single material print. Since Slic3r 1.38.3 the
two profiles behave the same in regard to single material print and
filament parking at the end of the print.

ir-fuel, 2018-01-03:
That's why. I always print in the "multimaterial" profile, even when using only 1 extruder (usually 3 or 4 because the way my printer is installed on my desk).