prusa3d/PrusaSlicer #664: Missing circle on first layer

rongith, 2018-01-04:
### Version
1.38.5

### Operating system type + version
Windows 10 x64

### Behavior
There is an inner 0.60mm wirth circle on the stl file that should be printed.
Expected:
[image]
Result:
[image]
Overview of the part:
[image]

#### STL/Config (.ZIP) where problem occurs
[01635-r02 - -Fini.zip](https://github.com/prusa3d/Slic3r/files/1603395/01635-r02.-.-Fini.zip)
[config.zip](https://github.com/prusa3d/Slic3r/files/1603397/config.zip)

bubnikv, 2018-01-04:
Do you have the elephant foot compensation enabled?
Do you have the 1st layer extrusion width set thicker than at the other
layers?
Do you have the "detect thin walls" feature disabled?

rongith, 2018-01-04:
Thank you for the very quick answer. The issue is elephant foot compensation.
Before posting, I tried to lower the perimeter width to 0.2mm without success and still saw a grey line but didn't thought to set elephant foot compensation back to 0.

closed 2018-01-04 by rongith

bubnikv, 2018-01-04:
I have an internal report to do something about the elephant foot
compensation, so this issue is not new to me.

rongith, 2018-01-04:
I closed because it seems to work as intended despite to have to be an expert to find the origin.
With 0.60mm part width, 0.49mm perimeter width and thin wall detection enabled, I can set a 0.20mm elephant foot compensation but not 0.25mm which is my default setting.