prusa3d/PrusaSlicer #665: Missing area to print.

rongith, 2018-01-04:
### Version
1.38.5 and earlier tested (1.34, .1.31)

### Operating system type + version
Windows 10 x64

### Behavior
There is some part missing between 40.00mm and 44.30mm on the inner surface. Moreover there are some top and bottom shells set at that place. Probably caused by top shells in the vicinity.
Expected:
[image]
Result:
[image]

#### STL/Config (.ZIP) where problem occurs
Same as https://github.com/prusa3d/Slic3r/issues/664
[01635-r02 - -Fini.zip](https://github.com/prusa3d/Slic3r/files/1603420/01635-r02.-.-Fini.zip)
[config.zip](https://github.com/prusa3d/Slic3r/files/1603422/config.zip)

rongith, 2018-01-04:
If the different circles on the base are linked by radial spokes, there is no issue.
[image]
[01635-r02 - -Imprime.zip](https://github.com/prusa3d/Slic3r/files/1603458/01635-r02.-.-Imprime.zip)

bubnikv, 2018-02-12:
When you merged the two parts with spokes, you likely forced your CAD to perform a boolean union. Without those spokes, you likely exported the two bodies as intersecting, but separate, and Slic3r does not like it.

In the future we would like to perform a boolean union slice by slice the same way Cura does it, but for now you better provide a valid model to Slic3r either by performing the boolean union in your CAD, or running your model through a STL fixing service.

FidelCapo, 2018-05-11:
Tested and fixed in Slic3rPE-1.40.0-alpha that will be released soon. Closing

closed 2018-05-11 by FidelCapo