prusa3d/PrusaSlicer #667: trying to print a Fusion 120mm x 100mm stl, slic3r says too big for my prusa mk2s, the bed is 250mmx210mm, it wants to scale it, don't want that

thecarpenteir, 2018-01-05:
### Version
_Version of Slic3r used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2018-02-12:
Would you please provide the STL? Thanks.

bubnikv, 2018-03-14:
Did not receive a response in two months, closing.

closed 2018-03-14 by bubnikv