prusa3d/PrusaSlicer #671: External perimeter quicker than inner

supermerill, 2018-01-06:
### Version
1.27.2
### Operating system type + version
windows 10

### Behavior
i can see that sometimes, the slicer put an higher speed on the external perimeter but my configuration tell him to do the opposite.

my configuration is below, as my stl.
when i slice it, it output this:

`
G1 Z0.600 F7200.000
;AFTER_LAYER_CHANGE
;0.6
G1 E-4.00000 F4800.00000
G1 Z1.100 F7200.000
G1 X131.878 Y113.378 F7200.000
G1 Z0.600 F7200.000
G1 E4.00000 F4800.00000
M204 S200
G1 F2203
G1 X111.122 Y113.378 E0.70256
G1 X111.122 Y103.622 E0.33022
G1 X113.940 Y103.622 E0.09538
G1 X114.041 Y104.585 E0.03279
G1 X114.534 Y106.102 E0.05396
G1 X115.331 Y107.482 E0.05396
G1 X116.398 Y108.667 E0.05396
G1 X117.687 Y109.604 E0.05396
G1 X119.144 Y110.252 E0.05396
G1 X120.703 Y110.584 E0.05396
G1 X122.297 Y110.584 E0.05396
G1 X123.856 Y110.252 E0.05396
G1 X125.313 Y109.604 E0.05396
G1 X126.602 Y108.667 E0.05396
G1 X127.669 Y107.482 E0.05396
G1 X128.466 Y106.102 E0.05396
G1 X128.959 Y104.585 E0.05396
G1 X129.060 Y103.622 E0.03279
G1 X131.878 Y103.622 E0.09538
G1 X131.878 Y113.318 E0.32819
M204 S1000
G1 X132.285 Y113.785 F7200.000
M204 S200
G1 F2400
G1 X110.715 Y113.785 E0.73012
G1 X110.715 Y103.215 E0.35778
G1 X114.306 Y103.215 E0.12157
G1 X114.442 Y104.500 E0.04375
G1 X114.908 Y105.935 E0.05106
G1 X115.662 Y107.242 E0.05106
G1 X116.671 Y108.363 E0.05106
G1 X117.892 Y109.249 E0.05106
G1 X119.270 Y109.863 E0.05106
G1 X120.746 Y110.177 E0.05106
G1 X122.254 Y110.177 E0.05106
G1 X123.730 Y109.863 E0.05106
G1 X125.108 Y109.249 E0.05106
G1 X126.329 Y108.363 E0.05106
G1 X127.338 Y107.242 E0.05106
G1 X128.092 Y105.935 E0.05106
G1 X128.558 Y104.500 E0.05106
G1 X128.694 Y103.215 E0.04375
G1 X132.285 Y103.215 E0.12157
G1 X132.285 Y113.725 E0.35575
M204 S1000`

here, we can see (if i'm not mistaken) that the external perimeter (the second one) is printed with a speed of 2400 and the inner one with a speed of 2203.

#### STL/Config (.ZIP) where problem occurs

https://drive.google.com/open?id=12ICkrpp5MsAMhgvKAbu9kA0DEvH11wCA

https://drive.google.com/open?id=14wbjBqwPZEBYWtM793JGG8jhlqPL61LB

florek41, 2018-01-14:
I have same issue on v.1.38.5 on windows 10
In my case part of the model prints OK (first ~50 layers), other part as @supermerill described.

Sebastianv650, 2018-01-14:
I bet that's not a bug, it's a feature. If the layer time is below min. layer time for cooling, slic3r slows down the print spead. But not speed of external perimeter as long as it's possible to reach min. time by adjusting other speeds, to keep the look of the external perimeter unchanged. Else you will see in the final print where the external perimeter was slowed down.

florek41, 2018-01-15:
@Sebastianv650 you have right about this. And now I wonder if we can get option to disable this feature. In my case, when printing overhangs this "cooling by speed" is crucial, so without slowing down outside perimeter the print curling up.

So it's kind of bug or unwanted future.

Sebastianv650, 2018-01-15:
I know what you mean, but would suggest another solution. Maybe your overhang is in a small layer so it's slowing down, but that's only by luck. 

See #242 for my proposal how overhang and print speed should be handled.

florek41, 2018-01-15:
Obviously You have right! I only had luck with those two factors. There is nothing to add :)

supermerill, 2018-01-15:
ok, so it's a duplicate of https://github.com/prusa3d/Slic3r/issues/536 and https://github.com/alexrj/Slic3r/issues/4070

closed 2018-01-21 by supermerill