prusa3d/PrusaSlicer #672: not responding in 3D TAB

Craftcadet, 2018-01-06:
### Version
_Version of Slic3r used goes here_
1.37.2 prusa3d. win64, I use multi material version and Mk3 prusa version. both versions respond the same
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 home,  V1709 OS build 16299.192 64bit 
### Behavior
* _Describe the problem_
not responding when in 3D TAB, works fine in 2D and preview. i have the latest Intel HD graphic driver - Installed Version: x.x.x.4653.  
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

supermerill, 2018-01-07:
see #233 #244 #268 #619 
it does the same on one of my pc. Have to reinstall same random intel drivers until it work again.

PlacidCat, 2018-01-27:
I have to add my voice to this as well. I've been using the latest stable version (1.38.5) and the 3D tab will simply lock for the most part. I've used Intel's Driver and Support Assistant and just updated my graphics driver (HD 4000) today (Jan. 26, 2018). Still the same issue. I've not seen anyone with a clear answer as to how to resolve this reliably. I have an MK3 on order and hoped to use Slic3r - I'd rather not resort to another slicer if possible.

supermerill, 2018-01-27:
You can try a older driver. There are no rule for this graphic driver bug.

PlacidCat, 2018-01-27:
I'll try that. If I find a version that works, I'll post it here. Thank you.

PlacidCat, 2018-01-27:
Well, that was surprisingly easy. There are only three versions of the Intel HD 4000 driver for Win 10 64-bit. Two I've already tried (the version I installed just yesterday and a version I had installed earlier in the month). The third is the earliest driver for Windows 10: 15.33.43.4425 (found [here](https://downloadcenter.intel.com/download/25977/Intel-Graphics-Driver-for-Windows-15-33-?product=81499)). Once I installed this and restarted the computer, Slic3r works beautifully. I hope this can be of some use to others.

Craftcadet, 2018-01-28:
Thanks PlacidCat! That worked for me too. thanks for link much appreciated.

PlacidCat, 2018-01-28:
Odd. When I started Slic3r this morning, the 3D view is once again slow. No updates or reboots happened in between. I'd be curious if others, once they update their driver, notice things getting bad again.

supermerill, 2018-01-28:
1) Windows 10 update his drivers without telling you.
or
2) this bug is really random and it reappear at every restart.

PlacidCat, 2018-01-28:
I'm guessing it might be 2. Intel still reports my drivers as being the same version as what I reported above. I might try rebooting and seeing what happens, just in case. I half wonder if there are conflicting libraries.

PlacidCat, 2018-01-28:
Rebooted. Slic3r works again. I can live with that, even if it is inconvenient. I *do* hope that Intel/MS sit down together and resolve this or, even better, for there to be some way for the Perl libraries to work around it.

bubnikv, 2018-01-28:
Guys, 

I am sorry to hear about the reappearing Intel HD Graphics issue. I am developing Slic3r on a laptop with the same class of an integrated graphics and I never had such issues.

Would you please provide as much as possible information about your hardware? What computer is that, what mark, what year, what exact setup etc. We may try to buy one of those to be able to find some workaround.

Thanks

PlacidCat, 2018-01-29:
Sorry for the delay. I've attached the dump from Intel's DSA software. Please let me know if you need any additional information. 

I appreciate your willingness to investigate this. I really feel the fault lies with the driver manufacturers and you are stuck with the mess.

[Detailed-System-Report.zip](https://github.com/prusa3d/Slic3r/files/1671996/Detailed-System-Report.zip)

PlacidCat, 2018-01-29:
On a side note, can you believe that even though Chrome is my default browser Windows decides to falsely claim IE is the default? :-)

supermerill, 2018-01-30:
Good pc : M-5Y51 (HD 5300) windows 10
Bad pc : i5-3317U (HD 4000) windows 10
i have an other one, but not in my house, i will edit this later.

justinvoelker, 2018-01-30:
Same issue here with HD 4000. Is there any way to just disable the 3D tab in the absence of a fix? I would even settle for showing a different tab by default so I can at least get the software started.

PlacidCat, 2018-01-30:
@justinvoelker - When I had the issue, I generally found that I can click one of the other tabs and hold the mouse button for a few seconds to ensure it registers and eventually that tab will appear. I often would select the Preview tab. After adding an object, a slicing of the model would, of course, cause it to appear.

If you have the time, try the link to the drivers I mention in a few posts back. It has worked well for me aside from a quirky instance where things slowed down again. I haven't seen that issue since, though and now I'm happily using the 3D tab.

bubnikv, 2018-01-30:
There is a parameter --no-plater for opening Slic3r just for editing the
parameters. This settings is used by the Repetier Host, but I doubt it will
be of much use to you.

I tried to work around this issue by providing an option to disable
anti-aliasing and OpenGL 2.0 features through a new "Use legacy OpenGL 1.1
rendering" checkbox at the preferences dialog. Would you please try this
option? Thanks.

bubnikv, 2018-01-30:
@PlacidCat Thanks for the report. Is it this laptop?
https://www.ebay.com/itm/Toshiba-Satellite-S855-S5378-15-6in-750GB-Intel-Core-i7-3rd-Gen-2-4GHz-8GB/152829928106?epid=129913810&hash=item23955fa2aa:g:z2YAAOSwHLNZVJcf

@supermerill 
> Bad pc : i5-3317U (HD 4000) windows 10

What exact brand and type of a laptop is it? I am trying to find something we can actually buy and I can debug on it.

Craftcadet, 2018-01-30:
My pc is the hp pavilion g6. Intel hd graphics 4000 driver.

PlacidCat, 2018-01-30:
@bubnikv Yep! That's it. I got mine for under $400 re-purposed. (Ex-business machine.) I would hope there are better deals out there. I should mention I tried the legacy mode in my search for results and had no luck. I read somewhere about the issue perhaps being due to frames not being "committed" at certain points (sorry, I don't remember the address or the specifics). I don't know if the rendering on the 3D tab and the slicing preview tab has a different pipeline but I certainly never had issues on the latter.

Thanks for looking into this.

supermerill, 2018-01-30:
It's an asus taichi 21 (ultrabook).
It doesn't work also on the cheap mini asus desktop computer my mom has. But i don't have the spec at hand.

justinvoelker, 2018-01-30:
Thanks @PlacidCat. I actually updated to the newest HD 4000 drivers [here](https://downloadcenter.intel.com/download/27417/Intel-Graphics-Driver-for-Windows-15-33) trying to fix an issue in with OpenGL 2.0 in OpenSCAD but it appears that the new drivers actually fixed this issue in Slic3rPE. So, I guess that's a possible fix others may want to try as well.

PlacidCat, 2018-01-31:
That's somewhat funny, @justinvoelker - those drivers didn't fix my issues with Slic3rPE. This, and the fact I had odd behavior even on my current drivers, really makes me wonder if there are two competing libraries on my computer. We have other machines in the house though, so I have a backup in case I can't get things working on mine. :-)

justinvoelker, 2018-01-31:
We'll it wouldn't be any fun if the problem were consistent! For the record, I'm on an ASUS S56CA-DH51.

bubnikv, 2018-01-31:
Anybody willing to part with his laptop configured to hang with Slic3r due to the Intel HD Graphics driver issues? Anybody to sell the laptop to Prusa Research for a reasonable price?

justinvoelker, 2018-01-31:
Not saying I wouldn't possibly part with my machine, but, is it really worth it to PR to buy my 5 year old laptop to debug an issue that is, at best, extremely sporadic?

bubnikv, 2018-02-01:
> Not saying I wouldn't possibly part with my machine, but, is it really
worth it to PR to buy my 5 year old laptop to debug an issue that is, at
best, extremely sporadic?

Unfortunately yes, it seems to be worth, as the issue happens too often to
be ignored, and without a way to reproduce the issue we will likely burn
much more money on developers trying to work around it blindly.

justinvoelker, 2018-02-01:
Ah, I see. Well @bubnikv, send me and email and we'll see if we can work something out.

roesel, 2018-02-16:
Same issue here. I'm running Windows 10 Pro on my Lenovo X1 Carbon 1st gen, Slicer for my brand new MK3 is useless, same as the MK2 version. What's funny is that the Prusa Control slicer works well. Technical details follow:

PC: Lenovo X1 Carbon 34447MG
CPU: Intel Core i5 3427U @ 1.80 GHz
GPU: Intel HD Graphics 4000
Driver:  10.18.10.4885 (latest)
Slic3r version: 1.38.6 (can't get to more because it's frozen)

(I'm also based in Prague so if there's any help I (or my laptop) could offer in person, let me know.)

supermerill, 2018-02-16:
Seems like it's happening on the HD4000 only.

justinvoelker, 2018-02-16:
It is very much Intel HD4000 related. It even sounds like different driver versions may or may not help. Drivers that did not work for others have worked for me and vice versa.

bubnikv, 2018-02-21:
@roesel It would help us immensely, if you could stop by with your laptop, so we could do couple of experiments. It would be great, if we could borrow your laptop for a day or two. We will happily compensate you with printing filament :-)

roesel, 2018-02-21:
@bubnikv I should be able to survive parting with my laptop for a few days. I could probably stop by your headquarters with the laptop and charger tomorrow morning on my way to work. Feel free to hit me up at roesel@gmail.com (in Czech if you want ;) ) and let me know if that works for you or if you want me to stop by a different day.

bubnikv, 2018-02-22:
Thanks to @roesel for lending his laptop to us for a day, I was able to investigate the issue and I believe I have a workaround for at least this particular laptop. We will release an updated beta today and we will hope for the best.

For the technical folks out there, the OpenGL driver hangs on the call of glFlush().

PlacidCat, 2018-02-22:
Funny enough, I had seen something online about this. Thanks @roesel and @bubnikv for helping move this forward.

PlacidCat, 2018-02-22:
By the way, my installation has decided to randomly act up again (no driver update - just stopped working) so I will test the commit as soon as there is a build. I will test after updating drivers as well.

Thanks everyone!

PlacidCat, 2018-02-22:
OK, I downloaded the two files and updated them locally. Works like a dream - immediately after updating the files, things started to work. I will update my drivers later and try again.

bubnikv, 2018-02-22:
If you have this issue, would you please re-test with the latest beta build and report the results?  Thank you.

https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1-beta1

The fix consists of removing the OpenGL glFlush() calls. The glFlush() call was superfluous, but it certainly should not lead to such adverse effects. Removal of glFlush() calls certainly solved the issue on one laptop, so let's hope it will fix all the other issues or at least most of them.

Thanks @roesel for providing a laptop for testing, it was a blessing to us.

closed 2018-02-22 by bubnikv

PlacidCat, 2018-02-22:
Worked with older drivers. Works with new drivers. I think we might have a winner!

Craftcadet, 2018-02-22:
GREAT! it worked for me too. Thank you very much for sorting this issue. Thank you @roesel  for the use of your laptop. im very happy now

wsmagee, 2018-02-26:
Fixed issue for me.