prusa3d/PrusaSlicer #673: Infill movements not complete in gcode

pigrew, 2018-01-06:
### Version
1.38.5-prusa3d-win64

### Operating system type + version
Windows10, x64

### Behavior
Exported g-code contains incomplete movements, so certain lines in the infill are truncated. The "Layers" preview looks correct, but the exported g-code is wrong (when viewed with "Pronterface" and when 3D-printed with a Marlin-based printer). I'm using 5% triangles for infill.

[image]

With the attached settings, I added the cylinder model, clicked slice, and then exported the gcode.
#### STL/Config (.ZIP) where problem occurs
[InfillProblem.zip](https://github.com/prusa3d/Slic3r/files/1609256/InfillProblem.zip)

bubnikv, 2018-01-06:
Are you sure it is not a pronterface issue?

pigrew, 2018-01-07:
I performed a 3D-print, and the artifacts where in the print.

bubnikv, 2018-01-07:
Please don't use the "Max volumetric slope positive/negative", leave these values at zero. This is an experimental feature and it seems to be broken.

pigrew, 2018-01-08:
I'll do that. Should we close this report?

bubnikv, 2018-02-02:
Let's leave it here, this behavior is something I have not seen yet.

bubnikv, 2019-01-29:
Disabled the experimental pressure equalizer with 57bd0889c384b4a2420adbfa306de8c8d191b9b4.
Closing.

closed 2019-01-29 by bubnikv