prusa3d/PrusaSlicer #676: Solid Layers count for Top is wrong, esp. when Ensure Vertical Shell is selected

asieving, 2018-01-08:
### Version
Version 1.37.2-prusa3d-win64

### Operating system type + version
Windows 7 Pro-64 bit SP1

### Behavior
Slicer does not generate the requested number of top solid layers.
In the supplied OBJ file, I selected 6 bottom layers, 6 top layers, and had "ensure vertical shell thickness" selected.  
The resulting file had 6 bottom layers and only 2 top layers.
It looks like this has had similar issues in the past (#60 and #79), but they have been marked as fixed?

I tried adjusting the bottom layer count, and that sometimes resulted in more top layers.  Also, I turned off Ensure vertical shell thickness, and that seemed to improve the top layer count.  I am attaching the OBJ file and the config file.  Slice it and observe the layers at 8.40mm .. 9.40mm
I believe, because of the chamber on the top edge, the top layers aren't all the same size, and so the vertical shell code is getting enforced, to the exclusion of the top layer count.  I think the correct behavior should be that top layer count would force the solid fill first, and then the vertical shell code would add in area, if it were still needed.
In this particular example, there is a solid layer at 8.40, then more infill layers, and then two layers of solid.  It should be solid from 8.40..9.40.

Let me know if you need more clarification.

#### STL/Config (.ZIP) where problem occurs

[qed-2018-0107-top-layers.zip](https://github.com/prusa3d/Slic3r/files/1610367/qed-2018-0107-top-layers.zip)

asieving, 2018-01-08:
I just now tried this on Version 1.38.6, and got the same result.

bubnikv, 2018-02-12:
Do I need to scale your object? Would you please scale your object for me, so I can reproduce your issue? Would you please add screenshots and possibly the G-code? Thanks.

asieving, 2018-02-12:
Hi, Thanks for looking at this issue.
I have attached a new ZIP file that should help illustrate the problem.

Starting with the input file  (either Base_input-issue-676-scale2540-rot90x .obj  or .stl
the next step is to scale the object by 2540, and then rotate on the X axis by 90.

Next, slice, (I am running Version 1.38.6-prusa3d-win64) I am using the configuration in the file
config-issue-676.ini

Observe the layers from 8.4mm to 9.4mm  These are the top 6 layers of a portion of the base.
Per the configuration, all six layers should be solid fill.  However,  8.4 is solid, then 8.6, 8.8, and 9.0 are back to the cubic infill, and then the final two layers, 9.2 and 9.4 are sold infill.  I've included screenshots of these 6 layers (n.m_is_[ok|wrong].png) as well as the layer immediately above, 9.6

I've attached the generated gcode  (Base_final-issue-676.gcode) and also observed an interesting fact:

If I export the STL file after scaling and rotating it, (Base_works-after-export.stl) then try loading and slicing this STL which Slic3r has exported, then I do not see the problem.  So, somehow there is something funky in the original STL file but the STL file created by Slic3r avoids the problem.

[issue-676-files.zip](https://github.com/prusa3d/Slic3r/files/1717021/issue-676-files.zip)

Thanks, and let me know if you need further information.

asieving, 2018-02-12:
sorry, hit close by accident.

bubnikv, 2018-02-12:
It is an effect of numerical instability of the underlying clipper library, when working with polygons with holes. I have patched some of the issues in #60 and #79, but not all of them. After I patched #60 and #79, I knew of only a single occurrence of this issue, and only on Linux. Now I have another example :-)

The issue could be worked around for this particular model by setting a resolution field at the Print Settings->Advanced->Resolution to 0.1mm. But I don't consider this advice to be a fix.

bubnikv, 2018-05-16:
Would you please retest with the Slic3r 1.40.0-alpha1? It seems this issue is no more reproducible with the alpha.

https://github.com/prusa3d/Slic3r/releases/tag/untagged-8fc8ca127d0924fe5504

asieving, 2018-05-16:
> 8fc8ca127d0924fe5504

Your download link gave me a 404, for some reason, but I got the
1.40.0-alpha1 build from github:
(
https://github.com/prusa3d/Slic3r/commit/3bc8d7517a0ff868b2a2731702872857dad01a44
)

https://github.com/prusa3d/Slic3r/releases/download/version_1.40.0-alpha1/Slic3rPE-1.40.0-alpha1.win64-full-201805141300.zip

It worked great with my model; the previous error is gone.

I really like what you've done with the presets and the configuration
pages.  Very nice, keep up the great work!

I think you can close my issue, with this new release.

Thanks and Best Regards-
Alan

bubnikv, 2018-05-16:
Thanks Alan for heads up. Our team appreciates the warm words.

I am not quite sure whether we really fixed this issue, or whether it has been just masked. We did some changes in the model import, which may or may not have fixed this issue. Anyway, as we cannot reproduce it with the current code base, let's close this issue. 

If you encounter a similar issue, please reopen this issue. Thanks.

closed 2018-05-16 by bubnikv